Ondsel-Development/FreeCAD #5: [Problem] FreeCAD needs a better way to manage custom data elements.

sliptonic, 2023-06-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
n/a
```

### Subproject(s) affected?

None

### Problem description

There are multiple competing solutions for users to create and maintain custom data properties.  There should be a single core feature that is universally used.  It should be accessible from both Python and C++ workbenches as well as providing strong support via the API.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2023-06-26:
https://ondsel.com/blog/build-system-for-custom-data-elements/

jedbrown, 2023-06-26:
I'm coming here after reading the blog and am interested in supporting design optimization in FEM workbench. For that, we need to identify a set of design parameters as being active, and will also need constraints. Some requirements/desirements from this perspective

1. Ability to apply non-exclusive labels (set membership) interactively and query all data elements in that set programmatically.
2. Ability to specify bound constraints (or more flexibly, a "prior" distribution, as in the [material overhaul proposal](https://forum.freecad.org/viewtopic.php?p=684108#p684108); a uniform distribution on an interval is equivalent to bound constraints).
3. Ability to apply/compare different solution values (e.g., produced by shape optimization with different objective functions).
4. (stretch) The constraint solver in sketcher knows the null space of the constraints in order to tell you how many degrees of freedom underconstrained your model is. It would be a uniquely powerful feature if one could ask the constraint solver to parametrize the null space into a labeled set that we could optimize over. Then instead of needing to tediously set and label constraints for a B-spline or complicated sketches, you could leave the model underconstrained and jump straight to optimization.

Related, but likely outside the scope of this proposal, we'd like to be able to evaluate the derivative of geometry elements with respect to parameters. In lieu of a more elegant approach, this could be done by finite differencing the model by programmatically changing parameter values (perhaps using the same interface as 3 without a graphical update).

julianstirling, 2023-07-14:
I have always disliked using the spreadsheet for variables. I think one core FreeCAD way to do variables is certainly the way forward for a cohesive user experience.

It is important (to me) that it is easy to me that it is intuitive to access the variables within sketcher functions. It would be nice if within a document I can set a length to:

    length-2*thickness

rather than something like

    doc.custom_data.properties.length-2*doc.custom_data.properties.thickness

The second is far less clear and user friendly. It also requires the user to remember the correct syntax/variable names to be able to get to the point where their own variables autocomplete.

Also it would be nice to be able to set a document level data from within sketcher. My idea workflow would be:

1. Open sketcher
2. Draw something
3. Constrain an element and set value as document level property from within constraint modal
4. Have constraint instantly accessible in any sketch with minimal syntax

AjinkyaDahale, 2023-07-14:
@julianstirling

> It is important (to me) that it is easy to me that it is intuitive to access the variables within sketcher functions. It would be nice if within a document I can set a length to:
>
> ``` [...]

That would be the aim, so long as the variable name (`length` or `thickness`in your above example) is not defined within the context. Even in that case, I expect the expression should look something like

```
CustomData.length - 2*thickness # Assuming length is defined in the sketch but thickness is not
```

> Also it would be nice to be able to set a document level data from within sketcher. My idea workflow would be:
>
> 1. Open sketcher
>
> 2. Draw something
>
> 3. Constrain an element and set value as document [...]

This can be achieved by adding a checkbox to the dialog setting a value. Then the constrain will be accessible using the aforementioned syntax.

julianstirling, 2023-07-14:
It would probably be a good idea to warn the user if the same name is used in document and an internal context? Otherwise it will probably cause some user confusion

AjinkyaDahale, 2023-07-14:
> It would probably be a good idea to warn the user if the same name is used in document and an internal context? Otherwise it will probably cause some user confusion. 

Good idea. Though I imagine it'd be easier if checking from the "internal context" (where we just need to check document-wide data) vs the other way (where we'd have to go through the variable lists of all existing objects).

carlopav, 2023-08-15:
I did read the article and really appreciated it. 
I just eventually wanted to add another possible existing solution that could have been missed: it's currently possible to add custom properties to a App::Part object and reference them inside the object tree (also inside sketches) using realthunder hidden reference (href()) and skip this way circular reference. This has the advantage of keeping the object parameters just inside the App::Part object itself. I have to say that using hidden reference is a bit annoying, but the result for the usecase of modifying an App::Part object is quite neat to me.
While writing I was also wondering if this custom data is expected to be document data or object data...

sliptonic, 2023-08-17:
> it's currently possible to add custom properties to a App::Part object and reference them inside the object tree (also inside sketches) using realthunder hidden reference (href()) and skip this way circular reference.

Can you point me to some documentation?

carlopav, 2023-08-17:
Hmmm, I'm not sure the workflow is documented... If I recall correctly,  it was introduced in main branch as part of @realthunder configuration table [PR](url=https://forum.freecad.org/viewtopic.php?p=400206#p400206).
I gave a quick look but i didn't find anything meaningful. Here you find a quick screenshot and a sample FC file showing the result. I hope it is on-topic with the discussion.

[image]

In short:
- create App::Part, a PartDesign::Body and a Sketch;
- select the App::Part, show all properties and add some custom properties in it, assign some values to them;
- now you can edit the sketch and you can reference the custom properties in the editor with "hiddenref(Part.CustomProperty)".
- in this way hiddenref() hides the reference from FreeCAD dependency checking.

[Hiddenref Example.zip](https://github.com/Ondsel-Development/FreeCAD/files/12370289/Hiddenref.Example.zip)

AjinkyaDahale, 2023-08-18:
> While writing I was also wondering if this custom data is expected to be document data or object data... 

This is supposed to be document data.

@carlopav could you elaborate where these circular references happen and how `huddenref` avoids them?

carlopav, 2023-08-18:
@AjinkyaDahale imagine: "Part.Group=[Sketch]", "Sketch.Constraint1=Part.Length". Unluckly I don't know what happens under the hood to skeep the circular reference.

pieterhijma, 2023-11-15:
I think this work has lots of potential!  I see opportunities in three categories:
- workflow
- properly defining parameters
- improved modular/parametric design

### Workflow

I mainly use the Spreadsheet workflow, but it is holding me back a great deal, especially if you like to work with many parameters and many different files.  I typically work with a "params.FCStd" file with only a spreadsheet called "params" that contains many of the parameters used in the design.  This setup (more on this later) requires me to type `<<params>>#<<params>>.my_var` all the time.  So, I fully agree with @JulianStirling's [comment](https://github.com/Ondsel-Development/FreeCAD/issues/5#issuecomment-1635575459).

Additionally, I keep on forgetting how I called the parameters exactly, I want to tweak the parameters, so I constantly go back and forth between the spreadsheet and the 3D viewer, something that really messes up the workflow.

### Properly defining parameters

I also feel the need to have different categories of parameters as I discuss [here](https://fabacademy.org/2023/labs/waag/students/pieter-hijma/week-05/index.html#design-and-3d).  In this design for a 3D-printed battery case I feel the need to specify "constants", for example the size of an AA or AAA battery, "parameters", for example that selects whether we model an AA or AAA battery case, "tolerances" for tolerances for a particular 3D printer, and "variables" which values I actually use in the design and is typically along the lines of `const_length_AA + tolerance_length`.  The [current PR](https://github.com/FreeCAD/FreeCAD/pull/10653) brings the notion of Groups that can help with this, although I think it is not really possible to actually do much with these Groups; they seem to only be a GUI concept.  Please correct me if I'm wrong.

### Suggestions 

#### Scopes / Context

As discussed [here](https://github.com/FreeCAD/FreeCAD/pull/10653#issuecomment-1812654735), the GUI automatically scopes the expressions for you.  I would like to see this changed, such that users can define their own scopes, enabling multiple scopes with checkboxes.  For example in my case, I would like to enable the scope `<<params>>#<<params>>`, such that `my_var` means the variable in that scope.

This allows the system to always understand what the fully qualified name is and it would give access to the document properties.  See [this comment](https://github.com/FreeCAD/FreeCAD/pull/10653#issuecomment-1812654735) for more information.

#### Defining variables

It would be great if we could have an expression `my_var(Tolerances)=length*2` that automatically defines a variable `my_var` in group "Tolerances" in the current scope.  I think a similar solution was already proposed above.

### Modularity/Parametric Design Problems

One of my annoyances is cyclic dependencies that break modularity.  I think the document-wide properties provide huge opportunities here.  Suppose you design something to be laser-cut and press-fit, for example my [Fab Academy final project](https://fabacademy.org/2023/labs/waag/students/pieter-hijma/final-project.html).

I have different components in different files, for example a track piece and a rail.  I laser cut that from the same material with a specific thickness and the slots for press-fitting depend on that thickness as well.  This means that there is a global parameter `thickness-wood` that should be used for both the track piece and for the rail.

One solution is to give both the track piece and the rail a parameter `thickness-wood` and let them have the same value.  The benefit of this is that both files are independent from each other, but the drawback is that if you want to change the thickness of the wood, you have to change it in all of the files.  It is better to keep to the [DRY](https://en.wikipedia.org/wiki/Don%27t_repeat_yourself) principle and make sure that if the thickness of the wood changes, both files follow automatically.

A better solution is to have the parameter `thickness-wood` in the main assembly file and have all parts refer to this parameter.  However, this is not possible, because with this, we create a circular dependency: the assembly file loads the track piece, and the track piece requires the assembly file.

My current non-satisfying solution is to have a separate `params.FCStd` file with only a spreadsheet called `params` that stores the parameters required for the design.  This file acts as a "sink" because all files are ultimately dependent on this file, the assembly, the track piece, and the rail.

However, this is terrible for modularity: if you want to exchange the track piece with a different project, you need to provide `params.FCStd` as well with all kinds of variables that may not be relevant.

### Suggestions for improving this

The current implementation provides a huge opportunity to improve on this!  Suppose the user is capable of making a group of properties called "Parameters" and the user is able to indicate that these parameters can be exposed to users of the file.  For example, both the rail and track piece have a group "Parameters" with variable `thickness-wood`.  These parameters have default values (to make sure the design is valid), but can also be set to user-defined values while opening a FreeCAD file, for example `FreeCAD --param thickness-wood=3 track-piece.FCStd`.

[... cut ...]  sets the parameter to the right value.

This gives us an elegant way to improve on modularity and parametric design.  In a sense, a FreeCAD file becomes a function (with an API that the user defines) that you can call with a set of parameters.  I don't know how other CAD programs solve this, but I think this will make FreeCAD really powerful.

Another suggestion is to add something as guards for these parameters, perhaps defined in the API that allow users to understand the range of parameters.

joshualibrarian, 2024-01-07:
> I just eventually wanted to add another possible existing solution that could have been missed: it's currently possible to add custom properties to a App::Part object and reference them inside the [...]

Hi!  Well, I also recently realized that one could add arbitrary properties to objects like the App::Part container.  My sense of organization really likes the idea of just having my part-wide parametric variables be properties of the part itself, but was frustrated by my inability to reference them there.  Your `hiddenref()` hack does indeed work, but I am really curious about the internals of WHY that reference isn't just available already and what we're doing with that work-around.

A also have been dis-satisfied with the Spreadsheet workflow (and the various other options) for my models, and really appreciated the Ondsel article linked above on the subject, which perfectly expresses the very issue I've been dealing with!  

I'm a developer who's been using FreeCAD a lot lately and am looking for how I can help make it better, and this issue is the one most relevant to the kind of modeling I do!  I also really appreciated your link above on variant links, as that may be close to something else I am trying to accomplish, something like "parameterized models" where I can design a model based on a handful of "parameters" and then generate more models automatically with variants of those parameters.

Anyway, I'm just trying to get oriented on the code base and where I can plug in, if anyone is actively working on this issue or has insight on the current state of it and the conversation around it, please let me know, I'm interested in helping.

pieterhijma, 2024-01-08:
Hi @joshualibrarian, thanks for your interest! I'm about to start working on this topic next week. My ideas have changed a bit since my last comments above and [on this PR](https://github.com/FreeCAD/FreeCAD/pull/10653#issuecomment-1812654735). I would like to find a solution that can be used FreeCAD-wide - also from C++ code - and improves on the following features:
- configuration tables (that require a hiddenref)
- variant links (make this FreeCAD-wide)
- solution for modularity as discussed above
- workflow by means of scopes as discussed above

The approach that I currently have in mind is creating a native property container modeled after Path PropertyBags, Assembly 4 Variables and Dynamic Data. Then this container has a property that allows users to expose the container to the user of the geometry. To create variant links, a user can import geometry into a model by means of a link but decide to use another copy of the exposed property container overriding this set of variables. This technique is inspired by the variant links that Assembly 4 introduced but makes it FreeCAD-wide. A similar technique can also be used to create configuration tables that is currently done with spreadsheets and hidden refs. In my humble opinion I find the use of hiddenrefs a bit problematic. I find that it makes the models a bit brittle, so I hope it is possible to achieve the above functionality without making use of hiddenrefs, most likely by making the property containers 'special' in the code.

For an improved workflow, I would like to have a way to declare a variable/parameter each time you are able to provide an expression. This variable/expression tuple is stored in the property container that has been marked as your current scope. A scope is simply a prefix of the whole path of the variable name, for example `myfile#mycontainer`. You could have the option to indicate one "write scope" where variables are stored and multiple "read scopes" from which you can reference variables in the expressions.

For my use case in FreeCAD, if the above would be reality, this would make my design process much more smooth. If I understand correctly from [your forum post](https://forum.freecad.org/viewtopic.php?t=84118) the same would be true for you. Please reach out to me for a discussion. I would love to discuss this further.

onekk, 2024-01-08:
Some consideration:

- FreeCAD have spreadhseet, that is intuitive to use but has some limitation like the fact that you have to create two tables to mixing input data and output data if they will end in objects. (There is for sure a better wording, I apologize)
- There are some already implemented solutions, requiring addons or "link Branch"

So probably there is no need as usual to reinvent the wheel. 

Even in this state you could make many things without hitting limitations, so some wiki pages that will extend maybe the "suggestion for robust model" will be probably a low hanging fruit.

It will be a good thing to start a "FreeCAD modelling guide" that will list all the "best practice" and "sane workflows" to make "robust models".

Based on these informations, that now are scattered around many forum posts and wiki pages, an user (both newbies and skilled users) could improve the way a model is created avoiding already know problems.

Then maybe during the process some decent idea could be discussed starting from the "state of the art".

I know that some things could be vanished until TNP mitigation was finished, but many of advices are not "TNP related" and finding them in an unique place should ease the learning curve for many users.

Sorry if this could seems OT.

Regards

Carlo D.

joshualibrarian, 2024-01-08:
> For my use case in FreeCAD, if the above would be reality, this would make my design process much more smooth. If I understand correctly from [your forum [...]

All those features sound good, I would very much like to continue this discussion, as variables like this are key to my workflow, and I have some other related features I want to implement, but at the moment I'm still at the "getting-to-know-the-codebase" stage.  It's been like an age of Middle Earth since I wrote any C++ code, but I do back-end Java for my day job, so it should come back quick enough.

A couple questions to get clear here... firstly, what the hell is `hiddenref()` for?  I notice that there is a `HiddenReference` object and a whole lotta references to it throughout the code base, lots in `Spreadsheet` and elsewhere, and with no particular comments to explain what it's about.  What is this class for, why would a reference be "hidden" or not and what does that mean, and why do we need a special method to access some references?  I would really appreciate some context here.

Secondly, when I (just the other day) learned (as you also mention) that any arbitrary object can have custom properties added to it, I thought this is ALMOST a solution, and I actually kinda like the idea of just improving that functionality (nesting, perhaps?) and just using it, but I would really want to put the values on the containing `App::Part` itself and that means understanding the above (about `hiddenref()`).  I can create an arbitrary object, like an `App::DocumentObjectGroup` and add properties to it and put it inside the `Part` and I can easily access it's properties... but not the properties of the containing `Part`!

The biggest issue I see with settling on using the object properties as data values is that (as pointed out by someone in that other thread) dependencies are recalculated on the object, not on parts of it (correct me if needed), so changing just one property would cause all dependent objects to recalculate just like in Spreadsheet, right?  I've reached a point with Spreadsheet where changing even an unreferenced text field hangs FreeCAD for like 15 minutes!

If we want each property to be an independent object, don't we really just need a simple dedicated class based on `DocumentObject`?  I am surprised to see that doing `App.activeDocument().addObject('App::DocumentObject', 'Something')` in the Python console generates a response like it worked, but doesn't actually create an object in the tree (like doing the same with `Part::Box` or `App::DocumentObjectGroup` does.  Why is this?  Is it just... abstract (or whatever C++ calls it)?

As to the question of scoping, I was actually kinda thinking that we could incorporate the use of `Part`, like... what if parts just all had properties, so if you are working on a feature inside of a part, it's properties could be in scope, and if you want to reference another part's properties it could be as easy as `file#PartWhatever.property`?  Is there some referencing issue why this won't work?  Again, it comes back to the `hiddenref` thing which I would like to dig into.

I have a lot to say about the whole variants thing too, but I think it best saved for another comment.  :)

pieterhijma, 2024-01-25:
Dear all, I created an [issue for this in the FreeCAD repository](https://github.com/FreeCAD/FreeCAD/issues/12120). I suggest to move the discussion of the problem there.

joshualibrarian, 2024-01-27:
@carlopav Well, as I get to know the codebase better, I'm throwing around some ideas for code improvements and improving `App:Part`, including to make VariantLinks more integral and easier to use.  Just wanted to let you know I did write a post, nearly two weeks ago, which currently has over 5500 views, but not a single reply.  I would be delighted to hear your feedback on it, even if you hate all my ideas.

https://forum.freecad.org/viewtopic.php?t=84334

closed 2025-01-01 by github-actions[bot]