Ondsel-Development/FreeCAD #19: [Problem] FreeCAD crashes on MacOS 13.6.3 w/ M1

brianredbeard, 2024-01-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Attempting to launch `Ondsel_ES_2024.1.0.35694-macOS-apple-silicon-arm64` results in a crash

### Full version info

```shell
Ondsel_ES_2024.1.0.35694 macOS apple silicon arm64
```

### Subproject(s) affected?

None

### Anything else?

[freecad_crash_report.log](https://github.com/Ondsel-Development/FreeCAD/files/14088868/freecad_crash_report.log)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

prokoudine, 2024-02-15:
Quick question: do you have the same issue with upstream FreeCAD build?

brianredbeard, 2024-02-16:
I do not.  Here's the about from upstream FreeCAD:

```
OS: macOS 13.6.4
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35822 (Git)
Build type: Release
Branch: main
Hash: c57a83cb4f7f8bc8840ebbfc9aba97c15a502bd9
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```

brianredbeard, 2024-03-15:
Since it's coming up on a few months since the last release can we expect a new release of Ondsel any time soon?

adrianinsaval, 2024-04-11:
does this happen with the development release of Ondsel? https://github.com/Ondsel-Development/FreeCAD/releases/tag/weekly-builds

closed 2024-12-30 by github-actions[bot]