Ondsel-Development/FreeCAD #23: [Problem] UI Errors on startup

30350n, 2024-02-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm getting a non usable UI and a bunch of errors on startup:
```
15:27:31  During initialization the error "UI file does not exist" occurred in C:\Programs\Ondsel\Ondsel_ES_2024.1.0.35694-Windows-x86_64\Mod\Arch\InitGui.py
15:27:31  Please look into the log file for further information
15:27:31  During initialization the error "UI file does not exist" occurred in C:\Programs\Ondsel\Ondsel_ES_2024.1.0.35694-Windows-x86_64\Mod\Draft\InitGui.py
15:27:31  Please look into the log file for further information
15:27:31  During initialization the error "Failed to wrap widget" occurred in C:\Programs\Ondsel\Ondsel_ES_2024.1.0.35694-Windows-x86_64\Mod\Ondsel-FirstRun\InitGui.py
15:27:31  Please look into the log file for further information
15:27:31  During initialization the error "Failed to wrap widget" occurred in C:\Programs\Ondsel\Ondsel_ES_2024.1.0.35694-Windows-x86_64\Mod\Ondsel-Lens\.\InitGui.py
15:27:31  Please look into the log file for further information
15:27:31  During initialization the error "Failed to wrap widget" occurred in C:\Programs\Ondsel\Ondsel_ES_2024.1.0.35694-Windows-x86_64\Mod\Tux\InitGui.py
15:27:31  Please look into the log file for further information
15:27:33  Access violation
15:27:33  QWidget::setMinimumSize: (qt_toolbar_ext_button/QToolBarExtension) Negative sizes (-18,0) are not possible
15:27:33  QWidget::setMinimumSize: (qt_toolbar_ext_button/QToolBarExtension) Negative sizes (-18,0) are not possible
15:27:33  QWidget::setMinimumSize: (qt_toolbar_ext_button/QToolBarExtension) Negative sizes (-18,0) are not possible
15:27:33  QWidget::setMinimumSize: (qt_toolbar_ext_button/QToolBarExtension) Negative sizes (-18,0) are not possible
15:27:33  QWidget::setMinimumSize: (qt_toolbar_ext_button/QToolBarExtension) Negative sizes (-18,0) are not possible
15:27:33  QWidget::setMinimumSize: (qt_toolbar_ext_button/QToolBarExtension) Negative sizes (-18,0) are not possible
15:27:33  QWidget::setMinimumSize: (qt_toolbar_ext_button/QToolBarExtension) Negative sizes (-18,0) are not possible
15:27:33  QWidget::setMinimumSize: (qt_toolbar_ext_button/QToolBarExtension) Negative sizes (-18,0) are not possible
15:27:33  QWidget::setMinimumSize: (qt_toolbar_ext_button/QToolBarExtension) Negative sizes (-18,0) are not possible
```
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of Ondsel: 64-bit
Version: 2024.1.0.35694 (Git)
Build type: Release
Branch: flavor
Hash: b070090c6873b3fcf4be9a459219d22edfc4e7cf
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/Germany (en_DE)
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2024-02-16:
Are you sure you correctly extracted everything? It would seem you are missing some files. Please try again with a fresh download

30350n, 2024-02-17:
> Are you sure you correctly extracted everything?

Yes.

> Please try again with a fresh download

Exact same issue.

closed 2024-12-10 by 30350n