Ondsel-Development/FreeCAD #25: [Problem] Appimage crashes on startup

Schildkroet, 2024-02-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Appimage crashes on startup:

```
Ondsel 2024.1.0, Libs: 2024.1.0R35694 (Git)
© FreeCAD team and Ondsel Inc., 2001-2024
Ondsel ES is based on the FreeCAD project and released under the terms of the LGPL2.1+ License.
Ondsel ES wouldn't be possible without the FreeCAD community.
      ###          #           #    ##### #####
     #   #         #           #    #     #
     #   # ###   ### #### #### #    #     #
     #   # #  # #  # #    #    #    ####  #####
     #   # #  # #  # #### ###  #    #         #
     #   # #  # #  #    # #    #    #         #
      ###  #  # #### #### #### ##   ##### #####

MESA-LOADER: failed to open radeonsi: /usr/lib/dri/radeonsi_dri.so: Kann die Shared-Object-Datei nicht öffnen: Datei oder Verzeichnis nicht gefunden (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: radeonsi
MESA-LOADER: failed to open zink: /usr/lib/dri/zink_dri.so: Kann die Shared-Object-Datei nicht öffnen: Datei oder Verzeichnis nicht gefunden (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: zink
MESA-LOADER: failed to open radeonsi: /usr/lib/dri/radeonsi_dri.so: Kann die Shared-Object-Datei nicht öffnen: Datei oder Verzeichnis nicht gefunden (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: radeonsi
MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: Kann die Shared-Object-Datei nicht öffnen: Datei oder Verzeichnis nicht gefunden (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: swrast
WebEngineContext used before QtWebEngine::initialize() or OpenGL context creation failed.
QGLXContext: Failed to create dummy context
Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples 0, swapBehavior QSurfaceFormat::DefaultSwapBehavior, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile) 
/tmp/.mount_Ondsel2NhFai/AppRun: Zeile 43: 21412 Abgebrochen             (Speicherabzug geschrieben) "${MAIN}" "$@"

```
[image]

### Full version info

```shell
Can't access it (Programm doesn't start)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

artelse, 2024-02-27:
Am having the same problem.
```
(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: iris
MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: swrast
WebEngineContext used before QtWebEngine::initialize() or OpenGL context creation failed.
QGLXContext: Failed to create dummy context
Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples 0, swapBehavior QSurfaceFormat::DefaultSwapBehavior, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile) 
/tmp/.mount_Ondsel4WBgx7/AppRun: line 43: 63570 Aborted                 (core dumped) "${MAIN}" "$@"
```

ShiroyukiX, 2024-02-28:
Same here. I recently updated my system for a reason unrelated to Ondsel/Freecad and the program crashed while in a project.

```Ondsel 2024.1.0, Libs: 2024.1.0R35694 (Git)
© FreeCAD team and Ondsel Inc., 2001-2024
Ondsel ES is based on the FreeCAD project and released under the terms of the LGPL2.1+ License.
Ondsel ES wouldn't be possible without the FreeCAD community.
      ###          #           #    ##### #####
     #   #         #           #    #     #
     #   # ###   ### #### #### #    #     #
     #   # #  # #  # #    #    #    ####  #####
     #   # #  # #  # #### ###  #    #         #
     #   # #  # #  #    # #    #    #         #
      ###  #  # #### #### #### ##   ##### #####

MESA-LOADER: failed to open radeonsi: /usr/lib/dri/radeonsi_dri.so: undefined symbol: amdgpu_va_get_start_addr (search paths /usr/lib/dri, suffix _dri)
failed to load driver: radeonsi
MESA-LOADER: failed to open radeonsi: /usr/lib/dri/radeonsi_dri.so: undefined symbol: amdgpu_va_get_start_addr (search paths /usr/lib/dri, suffix _dri)
failed to load driver: radeonsi
MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: undefined symbol: amdgpu_va_get_start_addr (search paths /usr/lib/dri, suffix _dri)
failed to load driver: swrast
QGLXContext: Failed to create dummy context
Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples 0, swapBehavior QSurfaceFormat::DefaultSwapBehavior, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile) 
/tmp/.mount_Ondselom8NR4/AppRun: line 43:  8085 Aborted                 (core dumped) "${MAIN}" "$@"
```

Schildkroet, 2024-02-28:
If i use `export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so` i get also the same error
`/tmp/.mount_OndseljWbBq3/usr/bin/freecad: symbol lookup error: /usr/lib/x86_64-linux-gnu/dri/radeonsi_dri.so: undefined symbol: amdgpu_va_get_start_addr`

stoky996, 2024-02-28:
weird i get a different error but on the same file here is mine

[image]

Schildkroet, 2024-02-28:
This command only applies to my Ubuntu System and tells Ondsel where to look for this driver. On your system it finds the driver.

stoky996, 2024-02-28:
adding the second line pulls up the same error

[image]

ShiroyukiX, 2024-02-29:
Downgrading `mesa` from `1:24.0.1-1` to `1:23.2.1-2` appears to have solved the issue for me. Looks like a recent update to the package is causing the issue.

This is my command output after the downgrade.
```
Ondsel 2024.1.0, Libs: 2024.1.0R35694 (Git)
© FreeCAD team and Ondsel Inc., 2001-2024
Ondsel ES is based on the FreeCAD project and released under the terms of the LGPL2.1+ License.
Ondsel ES wouldn't be possible without the FreeCAD community.
      ###          #           #    ##### #####
     #   #         #           #    #     #
     #   # ###   ### #### #### #    #     #
     #   # #  # #  # #    #    #    ####  #####
     #   # #  # #  # #### ###  #    #         #
     #   # #  # #  #    # #    #    #         #
      ###  #  # #### #### #### ##   ##### #####

/home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: No such file or directory

```
Links to info:
https://www.linuxquestions.org/questions/slackware-14/freecad-not-starting-4175733563/
https://community.clearlinux.org/t/mesa-fails-to-initialize/9386

Petapton, 2024-03-02:
> Downgrading `mesa` from `1:24.0.1-1` to `1:23.2.1-2` appears to have solved the issue for me. Looks like a recent update to the package is the cause.
> 
> This is my command output after the downgrade.
> 
> ```
> Ondsel 2024.1.0, Libs: 2024.1.0R35694 (Git)
> © FreeCAD team and Ondsel Inc., 2001-2024
> Ondsel ES is based on the FreeCAD project and released under the terms of the LGPL2.1+ License.
> Ondsel ES wouldn't be possible without the FreeCAD community.
>       ###          #           #    ##### #####
>      #   #         #           #    #     #
>      #   # ###   ### #### #### #    #     #
>      #   # #  # #  # #    #    #    ####  #####
>      #   # #  # #  # #### ###  #    #         #
>      #   # #  # #  #    # #    #    #         #
>       ###  #  # #### #### #### ##   ##### #####
> 
> /home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: No such file or directory
> ```
> 
> Links to info: https://www.linuxquestions.org/questions/slackware-14/freecad-not-starting-4175733563/ https://community.clearlinux.org/t/mesa-fails-to-initialize/9386

Thank you! It worked for me.
In case anyone was wondering, on Arch I used `sudo pacman -U https://archive.archlinux.org/packages/m/mesa/mesa-23.2.0-1-x86_64.pkg.tar.zst` to download an old version of mesa (https://wiki.archlinux.org/title/Arch_Linux_Archive#How_to_downgrade_one_package).
Please be cautious when downgrading packages.

brandonbeaumia, 2024-03-13:
Same issue here with the Ondsel 1.0 build, but not the 2024/03/11 git 36391 latest weekly FreeCAD AppImage. Hopefully the next build of Ondsel comes sooner than later since surely it will take in the base FC changes and work (hopefully not a build/CI issue?)

erikkallen, 2024-03-19:
I had the same problem and fixed it by rebuilding the appimage without libdrm
Can't find the forum post where I found this solution sorry, but the steps are below:

```
./Ondsel_ES_2024.1.0.35694-Linux-x86_64.AppImage --appimage-extract
rm squashfs-root/usr/lib/libdrm*
appimagetool -n squashfs-root/ Ondsel-2024.1-arch.AppImage
./Ondsel-2024.1-arch.AppImage
```

GreenCappuccino, 2024-04-29:
Are there plans to package Ondsel as a Flatpak? From what I understand it should bundle the mesa dependency so there wouldn't be an issue like this.

closed 2024-12-30 by github-actions[bot]