Ondsel-Development/FreeCAD #31: [Problem] missing approximate file sizes on main download page

Ryder17z, 2024-02-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you happen to be on a metered or otherwise limited connection, it can get frustrating to not immediately see the rough size of the file containing the software. This is especially true on flakey connections, like cellular networks.

### Full version info

```shell
N/A
```

### Subproject(s) affected?

None

### Anything else?

Additionally: a "subproject" for the website itself may be a good idea for other website related issues/suggestions/feedback.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

prokoudine, 2024-03-24:
That's a good point, thank you, @Ryder17z.

@JohnAD Let's discuss tomorrow

Ryder17z, 2024-12-09:
Shut up **stalebot & company**