Ondsel-Development/FreeCAD #34: Wrong top and botton plane.

dinosauria123, 2024-03-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I draw two over wrapped circles on sketcher and extrude solid them at the same, It shows strange rectangle planes at top and bottom of the objects.
[image]
[image]
[test.FCStd.zip](https://github.com/Ondsel-Development/FreeCAD/files/14595165/test.FCStd.zip)

### Full version info

```shell
OS: Ubuntu 23.10 (ubuntu:GNOME/ubuntu-wayland)
Word size of Ondsel: 64-bit
Version: 2024.2.0dev.36204 (Git) AppImage
Build type: Release
Branch: flavor
Hash: cfe8ba1084699057a1367965d315c118e4aabeeb
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Japanese/Japan (ja_JP)
Installed mods: 
  * Ondsel-Lens 2024.1.22.01
  * InventorLoader 1.4.0
  * sheetmetal 0.4.2
  * OpenDark 2023.12.17
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

prokoudine, 2024-04-07:
This isn't an Ondsel specific thing, it can be replicated in upstream FreeCAD as well. I think the program is expecting a kind of a sketch that can be extruded/padded into a solid body, but you are not giving it the right geometry. E.g. it needs to concentric circles to create a hollow cylinder. However, when you create offsets for the circles and trim intersection for them, it pads just fine. You'll also need a separate body for the cylinder that doesn't intersect other cylinders.

[2024-04-07-2.zip](https://github.com/Ondsel-Development/FreeCAD/files/14898720/2024-04-07-2.zip)

prokoudine, 2024-04-07:
BTW, Onshape has a problem interpreting the same case as well and does it differently, by summing two intersecting circles into one shape:

[image]

Unless, of course, this is how you were expecting Ondsel ES to behave :) In which case, please tell us what you expected to happen apart from not seeing the two planes.

dinosauria123, 2024-04-07:
Thank you for your comments. I hope this will be "Fixed".

dinosauria123, 2024-05-07:
Thank for implementing "Pad: Result has multiple solids: that is not currently supported." in Ondsel ES 2.0.

Stay tuned.

adrianinsaval, 2024-05-10:
"result has multiple solids" message exists for years now. This was either a bug in freecad's sketch profile validation that was fixed between then an now or this was an upstream opencascade bug that was fixed between 7.6.3 (used in previous release) and 7.7.2 (used now)

closed 2024-12-29 by github-actions[bot]