Ondsel-Development/FreeCAD #35: [Problem] Tree Covers Spreadsheet Cells

ahcox, 2024-03-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
The  Model Tree / Object Tree covers up spreadsheet cells. Tiling the windows if there is a simple part design window and one spreadsheet open puts the spreadsheet on the left under the tree.

### Full version info

```shell
OS: Ubuntu 23.10 (ubuntu:GNOME/ubuntu)
Word size of Ondsel: 64-bit
Version: 2024.1.0.35694 (Git) AppImage
Build type: Release
Branch: flavor
Hash: b070090c6873b3fcf4be9a459219d22edfc4e7cf
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * OpenDark 2023.12.17
  * Ondsel-Lens 2024.1.22.01
```
```

### Subproject(s) affected?

Spreadsheet

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

JohnOCFII, 2024-03-25:
Yes, I've seen that over the past few weeklies as well.  If I move my mouse to the right, the spreadsheet (properly) becomes the front "window" and the tree is hidden.

```
OS: macOS 14.4
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36526 (Git)
Build type: Release
Branch: main
Hash: 2ebbd83268ffabf41ceaad802714a4ffd9214437
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * OpenLight_Beta 2023.12.17
  * FeedsAndSpeeds 0.5.0
  * OpenDark 2023.12.17
```

[image]

kinghat, 2024-03-25:
my tree is auto-hiding. if i run my mouse over to the edge the tree expands properly:

[image]

[image]

```
OS: Fedora Linux 39 (KDE Plasma) (KDE/plasma)
Word size of Ondsel: 64-bit
Version: 2024.2.0dev.36532 (Git) AppImage
Build type: Release
Branch: flavor
Hash: ab2d6d4dee83f717c7e9ac68b89e620cfa26b974
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * OpenDark 2023.12.17
  * Ondsel-Lens 2024.1.22.01
  * CurvedShapes 1.0.8
  * Curves 0.6.29
  * sheetmetal 0.4.6
  * fasteners 0.5.15
  * DynamicData 2.60.0
  * freecad.gears 1.2.0
  * lattice2 1.0.0
  * IconThemes
```

JohnOCFII, 2024-03-26:
> my tree is auto-hiding. if i run my mouse over to the edge the tree expands properly:

Mine hides and unhides correctly AFTER the first initial display after I choose the spreadsheet (usually from the tab at the bottom of the window).  It is just the first time the spreadsheet shows up that the tree is still visible on top.

closed 2024-12-29 by github-actions[bot]