Ondsel-Development/FreeCAD #37: Gears workbench crashes ondsel

0phiTeS, 2024-03-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Ondsel crashes when trying to insert gears from the workbench.
Only crown, worm, lantern, hypocycloid, and timing seem to work.
Tried running ondsel through cmd with the -l argument -and the only error i get is: 

Err: Err: DeprecationWarningErr: : Err: Second argument is deprecated. It is ignored and will be removed in future versions. The default Python feature proxy is used for extension method overrides.Err: 
Err: Abnormal program termination...
Err: Terminating...

Tried running the workbench also on the stable release of ondsel and i get the same error/crash. 
I also run the workbench on freecad 0.22 and it works fine

### Full version info

```shell
OS: Windows 11 build 22621
Word size of Ondsel: 64-bit
Version: 2024.2.0dev.36532 (Git)
Build type: Release
Branch: flavor
Hash: ab2d6d4dee83f717c7e9ac68b89e620cfa26b974
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Greek/Greece (el_GR)
Installed mods: 
  * freecad.gears 1.0.0
  * Ondsel-Lens 2024.3.26.01
  * OpenDark 2023.12.17
  * sheetmetal 0.4.8
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

prokoudine, 2024-03-28:
> freecad.gears 1.0.0

Interesting. The up-to-date version of gears is 1.2.0 and it's been that way [for several months](https://github.com/looooo/freecad.gears/commit/6197bfdc16a927be15212aa8eed5b7fe9378e45d) already. There's something fishy going on and I'm not sure where.

I tried 1.2.0 on the same weekly build of OES that you have, except on Linux, and it doesn't crash.

@adrianinsaval Any ideas?

0phiTeS, 2024-03-29:
Maybe it has something to do with the fact that i am running it on my work pc? I don't know... I'll try it on my home pc and come back with an answer. That being said, it doesn't make sense that it works fine on the vanilla freecad and it doesn't on ondsel..

0phiTeS, 2024-03-29:
There is a similar issue that i just found out about, on the freecad.gears issue page: [](`https://github.com/looooo/freecad.gears/issues/156`)

closed 2024-12-29 by github-actions[bot]