Ondsel-Development/FreeCAD #42: TechDraw module causes reproducible exception and file corruption

memetb, 2024-05-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I open ondsel, create a part designer body, make a rectangle, pad it 10mm.

I then go to TechDraw workbench, select the Body object, switch over to Page tab, click the "Insert Projection Group" command, and immediately get the following exception:

```
20:12:57  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'Base.FreeCADError'>: vector::_M_range_check: __n (which is 18446744073709551615) >= this->size() (which is 3)
20:13:11  C++ exception thrown for 'TechDrawGui::DlgPrefsTechDrawAnnotationImp' (vector::_M_range_check: __n (which is 18446744073709551615) >= this->size() (which is 3))
20:14:32  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'Base.FreeCADError'>: vector::_M_range_check: __n (which is 18446744073709551615) >= this->size() (which is 3)
```

Strangely enough, I can still use tech draw, however saving a file in this state corrupts the project to the point that I can no longer open it in Ondsel and need to instead open it in FreeCad and delete the Page object.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/cinnamon)
Word size of Ondsel: 64-bit
Version: 2024.1.0.35694 (Git) AppImage
Build type: Release
Branch: flavor
Hash: b070090c6873b3fcf4be9a459219d22edfc4e7cf
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * fasteners 0.5.17
  * Ondsel-Lens 2024.4.16.01
  * Assembly4 0.50.12
  * OpenDark 2024.3.13
```
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

prokoudine, 2024-05-04:
Unfortunately, I can't replicate that with the same version on Ubuntu 23.10 (I don't have pure Debian around). Either way, several insertion commands have been merged into one for the coming new release, so let's revisit this soonish and see if the bug is still there in some form.

memetb, 2024-05-05:
Thank you @prokoudine. I will wait for the new release.

closed 2024-12-08 by memetb