Ondsel-Development/FreeCAD #43: 2024.2.0dev 37176: Startup Errors

casterle, 2024-05-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Upon opening the program I get the following warnings/errors when I start the program. I can't be sure they did not exist in a previous version; I haven't updated recently.

18:58:07  During initialization the error "module 'Utils' has no attribute 'getLogger'" occurred in C:\Users\caste\AppData\Roaming\Ondsel\Mod\Ondsel-Lens\.\InitGui.py
18:58:07  Please look into the log file for further information
18:58:38  QWindowsWindow::setGeometry: Unable to set geometry 1931x188-1013+771 (frame: 1947x227-1021+740) on QWidgetWindow/"Report viewWindow" on "\\.\DISPLAY8". Resulting geometry: 1101x105-1017+750 (frame: 1117x144-1025+719) margins: 8, 31, 8, 8 minimum size: 68x66 maximum size: 524287x524287 MINMAXINFO maxSize=0,0 maxpos=0,0 mintrack=84,105 maxtrack=524303,524326)

### Full version info

```shell
OS: Windows 11 build 22631
Word size of Ondsel: 64-bit
Version: 2024.2.0dev.37176 (Git)
Build type: Release
Branch: flavor
Hash: df5a9305b00cfa62e91ba9bb38772b1f157b6572
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.64
  * Assembly4 0.50.12
  * Behave-Dark-Colors 0.1.1
  * boltsfc 2022.11.5
  * CurvedShapes 1.0.9
  * Curves 0.6.33
  * Defeaturing 1.2.2
  * Dracula 0.0.5
  * DynamicData 2.62.0
  * ExplodedAssembly
  * fasteners 0.5.17
  * Freecad-Built-in-themes-beta 1.2.2
  * freecad.gears 1.2.0
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * MeshRemodel 1.9.23
  * Ondsel-Lens 2024.4.16.01
  * ProDarkThemePreferencePack 1.0.0
  * SelectorToolbar
  * Silk 0.1.5
  * toSketch 1.0.1
  * Trails 2022.1.0
  * woodworking 0.21.2.33771
  * workfeature
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

prokoudine, 2024-05-04:
@pieterhijma seems like it's up your alley, hence assigning to you

pieterhijma, 2024-05-09:
Dear @casterle, 

Thank you for reporting. I have been thinking about this for a while. I see that you have many addons. It could be the case that a different addon has a file `Utils.py` as well. Could you perhaps show me the output from these commands:

View -> Panels -> Python console

And then in the Python console, could you execute the following commands:

``` python
import Utils
Utils.__file__
```

The output that I would expect is:

```
<some path>\Ondsel\Mod\Ondsel-Lens\Utils.py
```

Thanks in advance.

casterle, 2024-05-09:
I’m glad to help Pieter,

 

Here’s what I get: 'C:\\Users\\caste\\AppData\\Roaming\\Ondsel\\Mod\\workfeature\\Utils\\__init__.py'

Everything (an outstanding, instant and free file search utility - voidtools.com) tells me I have 952 copies of that file, dated anywhere from a few days ago to some time in 2017! 

 

I’ve got FC 0.22dev, 0.21.2, and of course Ondsel weekly; I don’t know if that may be part of the issue.

I’m starting to learn Python so I can at least write FC extensions (long-retired dev here), which I’ve avoided partly due to what has caused me to have 952 copies of one file on my system.

 

BTW, I love what you folks at Ondsel are doing; as you’ve pointed out, it’s the best way forward for large open-source projects. I hope you guys get rich!

 

-Leroy

adrianinsaval, 2024-05-10:
It is not advised to use generic names such as Utils exactly due to this, I recommend renaming it to something like lens_utils to avoid conflicts

pieterhijma, 2024-05-10:
Dear Leroy,

Thank you for proving this information!

> I’m glad to help Pieter, 
> 
> Here’s what I get: 'C:\\Users\\caste\\AppData\\Roaming\\Ondsel\\Mod\\workfeature\\Utils\\__init__.py'

Yes, this means that the addon `workfeature` is in conflict with the Ondsel addon. While Ondsel is importing `Utils` it doesn't get the `Utils` that is part of the Ondsel addon, but the one from `workfeature`. A quick fix is to simply uninstall workfeature, but we'll improve on this as well.

> Everything (an outstanding, instant and free file search utility - voidtools.com) tells me I have 952 copies of that file, dated anywhere from a few days ago to some time in 2017!
> 
> I’ve got FC 0.22dev, 0.21.2, and of course Ondsel weekly; I don’t know if that may be part of the issue.

No, that is not the issue. The issue is that addon's typically don't use separate namespaces, so independent from us, there was a different addon, namely `workfeature` that has a file `Utils.py` as well (in this case it is a directory with an `init.py`). This will break either the Ondsel addon or the `workfeature` addon.

> I’m starting to learn Python so I can at least write FC extensions (long-retired dev here), which I’ve avoided partly due to what has caused me to have 952 copies of one file on my system.
>
> BTW, I [...]

Thank you, that's nice to hear!

I'll discuss this with my colleagues for the best way forward and I'll get back to you.

casterle, 2024-05-10:
Hello Pieter,

Thank you for your patient reply. I’ll uninstall workfeature, but I’m curious, is this something I installed? If so, it was because I was trying to keep FC and Ondsel in sync WB-wise. I’ve installed WB’s that I might be interested in so I don’t lose track of them, perhaps not the best idea! I like them in sync so if I run into an issue in one I can try it in the other. Are there any add-ons I should avoid in Ondsel?

closed 2024-12-28 by github-actions[bot]