Ondsel-Development/FreeCAD #44: 2024.2.0dev 37176" File menu missing?

casterle, 2024-05-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Just installed this version and can't find the menu bar? Alt-F opens the file menu dropdown, so I know the menu system is functional. I don't have any use for the scrolling workbench buttons that have apparently replaced it as I use the Selector Toolbar, which I prefer. Having no menu bar displayed means I have to press two keys every time I want to access something in the menu.

Was the menu bar removed on purpose? If so, please tell me how to get it back.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of Ondsel: 64-bit
Version: 2024.2.0dev.37176 (Git)
Build type: Release
Branch: flavor
Hash: df5a9305b00cfa62e91ba9bb38772b1f157b6572
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.64
  * Assembly4 0.50.12
  * Behave-Dark-Colors 0.1.1
  * boltsfc 2022.11.5
  * CurvedShapes 1.0.9
  * Curves 0.6.33
  * Defeaturing 1.2.2
  * Dracula 0.0.5
  * DynamicData 2.62.0
  * ExplodedAssembly
  * fasteners 0.5.17
  * Freecad-Built-in-themes-beta 1.2.2
  * freecad.gears 1.2.0
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * MeshRemodel 1.9.23
  * Ondsel-Lens 2024.4.16.01
  * ProDarkThemePreferencePack 1.0.0
  * SelectorToolbar
  * Silk 0.1.5
  * toSketch 1.0.1
  * Trails 2022.1.0
  * woodworking 0.21.2.33771
  * workfeature
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

casterle, 2024-05-04:
**To replace the workbench bar with the menu bar:** Workbenches/Workbench Type -> ComboBox

Perhaps this was mentioned on the Startup screen and I missed it. If not, it should be (please?).

kinghat, 2024-05-08:
once i removed all the appimages state files from .local/share/, .cache/ and .config/ it came back.

casterle, 2024-05-08:
> once i removed all the appimages state files from .local/share/, .cache/ and .config/ it came back. this will remove all your settings.

Thank you. I could have sworn I posted this as a reply to my message, but I did find a setting that restores the menu bar. In Preferences, go to Workbenches. Scroll down and change the Workbench Selector Type to ComboBox.

closed 2024-12-28 by github-actions[bot]