Ondsel-Development/FreeCAD #46: TechDraw: Dimension tool has huge offset

maxwxyz, 2024-05-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The dimension is created off screen (not visible) and when moving the mouse cursor you see that there is a huge offset between the cursor and the dimension. Maybe related to DPR as I have a two monitor setup, one with HiDPI, one with normal DPI (happens on both screens).

Demo:

https://github.com/Ondsel-Development/FreeCAD/assets/6246609/735d4ab7-ad60-4abe-a8ba-4d8c353dfadd

### Full version info

```shell
OS: Windows 11 build 22631
Word size of Ondsel: 64-bit
Version: 2024.2.0dev.37179 (Git)
Build type: Release
Branch: flavor
Hash: 5d0133c096bfd2ff009350bc8dc1b9363542e473
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Ondsel-Lens 2024.1.22.01
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-05-10:
Have you tried when only one screen is connected and if changing the resolution?

maxwxyz, 2024-05-11:
It works when I only use one screen but I think this is not ideal to disconnect the laptop when using FreeCAD

PaddleStroke, 2024-05-13:
Of course. I'm just trying to pin-point the case where it doesn't work.

PaddleStroke, 2024-05-13:
I have tested on a dual screen setup and I cannot reproduce the issue. Here is my setup : 
- windows 10
Laptop screen : 3000x2000 (200%)
main screen : 3800x1600 (125%)

I have tested with various resolution and scale and cannot find a similar case as you. Can you please try to pinpoint exactly when it happens?

PaddleStroke, 2024-05-13:
Bug fixed in https://github.com/FreeCAD/FreeCAD/pull/13525

maxwxyz, 2024-05-13:
ah this could be, I've tested with one monitor on a simple view ... sorry!

closed 2024-12-08 by maxwxyz