Ondsel-Development/FreeCAD #52: OSX Windows Appear Offscreen

stevenwalton, 2024-05-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When opening menus windows can appear offscreen. Interestingly, when starting this issue I found that this will even happen if I switch windows and it stays on top until I click exit. **This is not a new issue to this new version.** It happened on the previous version as well, I just didn't report.

I think this happens because Macs have this annoying property where you can pull windows off screen. To fix new windows should probably appear within main Ondsel window context. There are some other minor annoying quality of life issues as well like being able to pick points.

The screenshot isn't particularly egregious but sometimes this window barely appears on screen
[image]

Kinda funny, probably a hint at the issue
[image]

Overall great work and I'm happy you guys are doing this! I especially like the new VarSets. What a big improvement over the annoying spreadsheets and macro :/ It can be a little confusing at first and I hope the UI/UX changes a little, but overall this is such a quality of life improvement! 

### Full version info

```shell
OS: macOS 14.4.1
Word size of Ondsel: 64-bit
Version: 2024.2.0.37191 (Git)
Build type: Release
Branch: (HEAD detached at 2024.2.0)
Hash: 2ad5fb327d068ae6280cf702d528b3a6e35770a7
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * Ondsel-Lens 2024.4.16.01
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2024-05-24:
actually this seems to be a duplicate of #51

adrianinsaval, 2024-05-24:
this might be a general problem of the expression dialog not macos specific as I think I experienced this on linux too

> I think this happens because Macs have this annoying property where you can pull windows off screen.

I could be wrong but I think all OSs allow this

stevenwalton, 2024-05-24:
I'll test on Linux this weekend. I have a recent install of EndeavourOS, though I was having difficulties building from source. I think I have time this weekend to give it another go.  

> I could be wrong but I think all OSs allow this

I don't think this is default behavior but I avoid windows machines and reluctantly use macs lol. But with nix, it could depend on the window manager and settings. I think the solution might be similar. If you point me in the right direction I might be able to give a hand. 

> actually this seems to be a duplicate of https://github.com/Ondsel-Development/FreeCAD/issues/51

Looks like a collision. These are only a few hours apart and I did not actually see the issue when I wrote mine (I did search). I'm not sure how you like to handle merging of issues but feel free to squash this issue and I'll subscribe to #51

closed 2024-12-28 by github-actions[bot]