Ondsel-Development/FreeCAD #53: Ondsel CFDOF Material Manager has no attribute

lt72884, 2024-05-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After installing the CFDOF workbench into Ondsel, it always crashes when trying to select the fluid material property such as air, water, etc.  It has been verified that CFDOF works in 0.22dev version and the current weekly builds. Most likely Ondsel is built on an older version of 0.22dev before the cfdof fixes

file “C:\Users\Matt\AppData\Roaming\Ondsel\Mod\CfdOF.\CfdOF\Solve\CfdFluidMaterial.py”, line 149, in setEdit
self.taskd = TaskPanelCfdFluidProperties.TaskPanelCfdFluidProperties(self.Object, physicsmodel)
File “C:\Users\Matt\AppData\Roaming\Ondsel\Mod\CfdOF.\CfdOF\Solve\TaskPanelCfdFluidProperties.py”, line 61, in _init
self.populateMaterialsList()
File “C:\Users\Matt\AppData\Roaming\Ondsel\Mod\CfdOF.\CfdOF\Solve\TaskPanelCfdFluidProperties.py”, line 126, in populateMaterialsList
self.materials, material_name_path_list = CfdTools.importMaterials()
File “C:\Users\Matt\AppData\Roaming\Ondsel\Mod\CfdOF.\CfdOF\CfdTools.py”, line 1626, in importMaterials
material_name_path_list = material_name_path_list + addMatDir(system_mat_dir, materials)
File “C:\Users\Matt\AppData\Roaming\Ondsel\Mod\CfdOF.\CfdOF\CfdTools.py”, line 1638, in addMatDir
materials[a_path] = importFCMat.read(a_path)
File “D:\Cloud Master\Ondsel ES\Mod\Material\importFCMat.py”, line 102, in read
materialManager = Material.MaterialManager()
<class ‘AttributeError’>: module ‘Material’ has no attribute ‘MaterialManager’

### Full version info

```shell
OS: Windows 10 build 19045
Word size of Ondsel: 64-bit
Version: 2024.1.0.35694 (Git)
Build type: Release
Branch: flavor
Hash: b070090c6873b3fcf4be9a459219d22edfc4e7cf
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Ondsel-Lens 2024.3.26.01
  * OpenDark 2023.12.17
```
```

### Subproject(s) affected?

None

### Anything else?

file “C:\Users\Matt\AppData\Roaming\Ondsel\Mod\CfdOF.\CfdOF\Solve\CfdFluidMaterial.py”, line 149, in setEdit
self.taskd = TaskPanelCfdFluidProperties.TaskPanelCfdFluidProperties(self.Object, physicsmodel)
File “C:\Users\Matt\AppData\Roaming\Ondsel\Mod\CfdOF.\CfdOF\Solve\TaskPanelCfdFluidProperties.py”, line 61, in _init
self.populateMaterialsList()
File “C:\Users\Matt\AppData\Roaming\Ondsel\Mod\CfdOF.\CfdOF\Solve\TaskPanelCfdFluidProperties.py”, line 126, in populateMaterialsList
self.materials, material_name_path_list = CfdTools.importMaterials()
File “C:\Users\Matt\AppData\Roaming\Ondsel\Mod\CfdOF.\CfdOF\CfdTools.py”, line 1626, in importMaterials
material_name_path_list = material_name_path_list + addMatDir(system_mat_dir, materials)
File “C:\Users\Matt\AppData\Roaming\Ondsel\Mod\CfdOF.\CfdOF\CfdTools.py”, line 1638, in addMatDir
materials[a_path] = importFCMat.read(a_path)
File “D:\Cloud Master\Ondsel ES\Mod\Material\importFCMat.py”, line 102, in read
materialManager = Material.MaterialManager()
<class ‘AttributeError’>: module ‘Material’ has no attribute ‘MaterialManager’

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

lt72884, 2024-05-24:
> I am unable to reproduce this with the latest version of cfdof, are you
> still experiencing this? can you give me step by step reproduction steps?

It was working in 2024.1 release due to materials not ported over, but it
seems that 2024.2, they got it figured out... I hope.

To reproduce the issue, just double click on fluid properties and it will
error out with the error i posted. But that was in 2024.1 ondsel

adrianinsaval, 2024-05-24:
ah ok since this was posted around the time we released 2024.2 I thought it had been reported for it. Indeed it works in 2024.2 so we can close this

closed 2024-05-24 by adrianinsaval