Ondsel-Development/FreeCAD #54: An error message is reported when starting on the Manjaro system

chwt163, 2024-05-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Ondsel can be run, but it stutters and often becomes unresponsive.

`Ondsel 2024.2.0, Libs: 2024.2.0R37191 (Git)
© FreeCAD team and Ondsel Inc., 2001-2024
Ondsel ES is based on the FreeCAD project and released under the terms of the LGPL2.1+ License.
Ondsel ES wouldn't be possible without the FreeCAD community.
      ###          #           #    ##### #####
     #   #         #           #    #     #
     #   # ###   ### #### #### #    #     #
     #   # #  # #  # #    #    #    ####  #####
     #   # #  # #  # #### ###  #    #         #
     #   # #  # #  #    # #    #    #         #
      ###  #  # #### #### #### ##   ##### #####

(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
QFont::fromString: Invalid description 'HarmonyOS Sans SC,10,-1,5,400,0,0,0,0,0,0,0,0,0,0,1'
QFont::fromString: Invalid description 'HarmonyOS Sans SC,10,-1,5,400,0,0,0,0,0,0,0,0,0,0,1'
QFont::fromString: Invalid description 'HarmonyOS Sans SC,10,-1,5,400,0,0,0,0,0,0,0,0,0,0,1'
During initialization the error "HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Max retries exceeded with url: /Ondsel-Development/Ondsel-Lens/master/package.xml (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x73b9ad511590>: Failed to establish a new connection: [Errno 111] 连接被拒绝'))" occurred in /home/chen/.local/share/Ondsel/Mod/Ondsel-Lens/./InitGui.py
Please look into the log file for further information
      % `

[image]

### Full version info

```shell
OS: Manjaro Linux (KDE/plasmawayland)
Word size of Ondsel: 64-bit
Version: 2024.2.0.37191 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 2024.2.0)
Hash: 2ad5fb327d068ae6280cf702d528b3a6e35770a7
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Chinese/China (zh_CN)
Installed mods: 
  * Ondsel-Lens 2024.5.6.01
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

prokoudine, 2024-05-17:
@pieterhijma Do you have ideas what's going on here?

pieterhijma, 2024-05-17:
Thank you @prokoudine for letting me know and thank you @chwt163 for reporting. I will look into it.

pieterhijma, 2024-05-20:
@chwt163. I looked into it. It turns out that the addon didn't handle exceptions well in the above case. I created the issue https://github.com/Ondsel-Development/Ondsel-Lens/issues/101 and submitted a PR. When merged, the addon will notify you for an update.

The max retries error shows when the Ondsel Lens addon cannot get a connection for a version check which can be caused by many issues. Since a version check is not critical, the version check silently fails if github refuses a connection or if there is a different connection error. The "silently" part was failing, so I improved the silent part by handling exceptions better.

- The other message you get is not something I cannot solve, perhaps @adrianinsaval has an idea?
- The stuttering and unresponsiveness may have to do with an unreliable connection. In that case, I would suggest to log out of the Ondsel Lens addon. Could you please report back if the stuttering and unresponsiveness disappears in that case?

Again, thanks for reporting!

chwt163, 2024-05-20:
Thank you so much! If it's fixed, can you notify the lssues here?

adrianinsaval, 2024-05-24:
> HarmonyOS Sans

can you search for this string at ~/.config/Ondsel/user.cfg and tell me in what parameters this is set? Probably is a good idea to remove those from your config as it seems to be an invalid font that was somehow configured for you

> (qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""

unfortunately we don't have good wayland support yet, maybe this will improve once we move to qt6

chwt163, 2024-05-25:
> can you search for this string at ~/.config/Ondsel/user.cfg and tell me in what parameters this is set? Probably is a good idea to remove those from your config as it seems to be an invalid font [...]

After changing to the system's built-in font, the startup prompt is only this: (qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
But the operation is still stuck, I tried the original freecad, and it is also running stuttering

[image]

adrianinsaval, 2024-05-27:
The stuttering problem seems to be a problem in qt/kde, people have reported that it's fixed with the lates update in archlinux to qt 6.7.1 and kde 6.0.5, these updates should reach manajaro's repos in a few weeks

closed 2024-12-07 by chwt163