Ondsel-Development/FreeCAD #56: [PROBLEM] 3DConnection SpaceMouse Wireless

microinteractive, 2024-05-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi,
SpaceMouse Wireless not showing in menu Tools -> Customize -> Spaceball Motion & Spaceball Buttons
it says "No Spaceball Present" and can't change settings for it.

[image]

### Full version info

```shell
ONDSEL:
OS: Windows 10 build 19045
Word size of Ondsel: 64-bit
Version: 2024.2.0dev.37179 (Git)
Build type: Release
Branch: flavor
Hash: 5d0133c096bfd2ff009350bc8dc1b9363542e473
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Installed mods: 
  * fasteners 0.5.20
  * freecad.gears 1.2.0
  * Ondsel-Lens 2024.5.6.01

3DConnection Driver:
3DxWare 10.8.20.3729
3DxWinCore 17.8.20.20889
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

microinteractive, 2024-05-25:
> does it show on freecad? is your mouse still supported by 3dconexxion? do you have the latest driver installed?

ONDSEL 2024.1 - showing, working, can adjust
ONDSEL 2024.2 - not showing, working, can't adjust 

FreeCad 0.21.2 - showing, working, can adjust

Mouse is still full supported.

Installed latest drivers:
3DConnection Driver:
3DxWare 10.8.20.3729
3DxWinCore 17.8.20.20889

prokoudine, 2024-05-25:
This is because 3Dconnexion developer changed this after Ondsel ES 2024.1 and FreeCad 0.21.2 releases, but before the Ondsel ES 2024.2 release.

prokoudine, 2024-05-25:
@pskowronskiTDx Could you please clarify what users should be doing now?

pskowronskiTDx, 2024-05-27:
The Spaceball Motion and Spaceball Buttons tabs are going to be removed soon. In order to change the device settings now, please use 3DxWare SmartUI ("Advanced settings" button).

microinteractive, 2024-05-27:
> The Spaceball Motion and Spaceball Buttons tabs are going to be removed soon. In order to change the device settings now, please use 3DxWare SmartUI ("Advanced settings" button).

Ok but Ondsel don't take any of settings from 3DxWare You can move sensitivity to min left  and still it flyes like crazy.
Same situation is with revese direction tick - You can mark/unmark no affect to movment in Onsel

Only 3DxWare detacts that is FreeCad

pskowronskiTDx, 2024-05-27:
I think you are running into this issue: https://github.com/FreeCAD/FreeCAD/issues/14044

microinteractive, 2024-05-27:
> I think you are running into this issue: [FreeCAD#14044](https://github.com/FreeCAD/FreeCAD/issues/14044) . I have provided the temporary solution there. We are currently working on separate configuration file for the new integration.

Yes this helps for Me
now i can do settings from 3DxWare and have more options in 3DxWare

closed 2024-05-28 by microinteractive