Ondsel-Development/FreeCAD #61: setting invalid values in the cylindrical constraint limits causes all the limits to silently fail

kevenwyld, 2024-05-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When attempting to set limits for the cylindrical constraint only certain values for min and max seem to be valid.  Negative values for min, positive values for max, etc.  When setting a value that is invalid, for example a positive min, all the limits silently break.  There doesn't seem to be any error output visible to the user in the console or report view, the object is just allowed to move past all the limits when any one of the values is invalid.

Thanks!

### Full version info

```shell
OS: Arch Linux (sway/sway)
Word size of Ondsel: 64-bit
Version: 2024.2.0.37191 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 2024.2.0)
Hash: 2ad5fb327d068ae6280cf702d528b3a6e35770a7
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Ondsel-Lens 2024.5.6.01
```

### Subproject(s) affected?

Assembly

### Anything else?

https://github.com/Ondsel-Development/FreeCAD/assets/11936535/7619959a-e542-4b7e-bafa-8fbe29829f93

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-05-30:
Thanks for the report.
Note that limits have merged into freecad main. So it is best to post the issue there, so that more people can see it.

Onto the problem itself: We could add a check on joint creation to check if the limits are valid and prevent validation if they are not.

PaddleStroke, 2024-05-30:
The alternate solution is to fix automatically if the limits are invalid. So if min is less than max we just swap them when solving.
This way we avoid the blocking error message.

kevenwyld, 2024-06-03:
Thanks @PaddleStroke!  I'll try to test the PR you linked if I can manage to build it.

closed 2024-06-04 by PaddleStroke