Ondsel-Development/FreeCAD #77: [1.0 bughunt] When nothing is selected, a delete command produces misleading error message

raindropsfromsky, 2024-06-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I click in space, and then press **CTRL+A**, Ondsel deselects everything.

At this stage, if I hit **Del**, Ondsel produces this error:
`ViewProviderSketch: Delete: Selection not restricted to one sketch and its subelements`

**Desired behavior:** 
It should not produce any error at all, because the **Delete** command did _not_ fail.
(There was no selection to be deleted, and so nothing was deleted.) 

Secondly, the error implies that I have selected something outside one sketch, which is completely misleading!
(An error message must guide the user how to correct his mistake.) 

### Full version info

```shell
OS: Windows 11 build 22631
Word size of Ondsel: 64-bit
Version: 2024.2.0dev.37608 (Git)
Build type: Release
Branch: flavor
Hash: cbb758c5dc79fd4c016a9cd45a007d3c7ce1a689
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/India (en_IN)
Installed mods: 
  * Ondsel-Lens 2024.3.26.01
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2024-06-13:
Good bug.  
This happens when editing a sketch.  
Make two different sketches.  Make both visible and start editing one of them.
You can't select edges from the other.  This makes sense.  

But you can use Ctrl-A to select the other one.  Then deleting gives the error.  Selection should be limited to only the currently edited sketch.

raindropsfromsky, 2024-06-14:
BTW, I had created only one sketch in the file. 
I had started with Part Design WB, and I had opened that sketch for editing in the Sketcher WB. 
That's when I saw the error.

Even if the body has multiple sketches, CTRL+A selects all sketches only in Part Design WB; not when we are in the Sketcher WB!
The Sketcher WB opens one sketch at a time, and it does not have access to the other sketches in the Part Design WB.
So a CTRL+A in the Sketcher WB will not select the other sketches.

Therefore, we will never encounter the condition reflected by this error, IMHO.

closed 2025-02-11 by github-actions[bot]