Ondsel-Development/FreeCAD #78: [1.0 bughunt] While creating a polyline, if I double-click, an error is produced

raindropsfromsky, 2024-06-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While creating a polyline, if I double-click, then an error dialog pops up: 

```
Failed to add line
[OK]
```

The error log is:
```
pyException: Traceback (most recent call last):
 File "<string>", line 1, in <module>
<class 'Part.OCCError'>: Both points are equal

Unnamed#Sketch: App.getDocument('Unnamed').getObject('Sketch').addGeometry(Part.LineSegment(App.Vector(-20.000000,-16.666592,0),App.Vector(-20.000000,-16.666592,0)),False)

Unnamed#Sketch: Failed to add line
```

**Desired behavior:**
1. Instead of producing an error, Ondsel should just finish the multiline.
2. Further, the _Polyline_ tool should remain selected, so that the user can start drawing the next polyline.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of Ondsel: 64-bit
Version: 2024.2.0dev.37608 (Git)
Build type: Release
Branch: flavor
Hash: cbb758c5dc79fd4c016a9cd45a007d3c7ce1a689
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/India (en_IN)
Installed mods: 
  * Ondsel-Lens 2024.3.26.01
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2024-06-13:
Good bug.   A bit hard to replicate.

You need to be drawing a polyline and doubleclick at the endpoint.  If the mouse moves at all, it works correctly and starts either adds a new segment or finishes the old one.

PaddleStroke, 2024-08-30:
Polyline needs to be rewriten completely to use the new handler framework (the one with the on-view-parameters).
Btw there is an issue in upstream already : https://github.com/FreeCAD/FreeCAD/issues/13924

PaddleStroke, 2024-08-30:
I'll close as upstream already has this issue.

closed 2024-08-30 by PaddleStroke