Ondsel-Development/FreeCAD #84: [1.0 Bughunt] Draft command breaks when the angle is increased beyond a random value; Ondsel becomes unresponsive

raindropsfromsky, 2024-06-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create a sketch as shown 
[image]
(Take care to keep the center of the circle outside the rectangle)
2. Close the sketch and pad it.
[image]
3. Apply Draft to the top face:
[image]
4. Increase the draft angle. At some point, the command breaks.
[image]
Ondsel becomes unresponsive (a rotating circle appears on the Draft panel)
Sometimes, the UI grays out and the top bar of Ondsel says not responding (but this is less frequent).

I could not find when the command breaks. There seems to be no pattern.
For example, at that angle, the geometry has NOT become impossible to solve.

**Desired behavior:**
The draft should not break at certain angle.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of Ondsel: 64-bit
Version: 2024.2.0dev.37608 (Git)
Build type: Release
Branch: flavor
Hash: cbb758c5dc79fd4c016a9cd45a007d3c7ce1a689
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/India (en_IN)
Installed mods: 
  * Ondsel-Lens 2024.3.26.01
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

raindropsfromsky, 2024-06-13:
I repeated the experiment by deleting the draft, and trimming the inside edges of the circle and the rectangle.
[image]

Then I applied a fresh draft. 

But when I increased the draft angle, it caused a worse effect on the geometry!
1. The draft function broke
2. All the trimmed edges reappeared!
3. The top surface has got detached from the rest of geometry

[image]

sliptonic, 2024-06-13:
confirmed.  For me, it breaks at exactly 32 degrees.  Reversing pull direction and it breaks at 58. 
it doesn't throw any red in the console so I'm not tagging it for the bug hunt.   It may well be an OCCT bug.  Needs more investigation.

closed 2025-02-11 by github-actions[bot]