Ondsel-Development/FreeCAD #85: [1.0 Bughunt] Unpredictable result with overlapping shapes

raindropsfromsky, 2024-06-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the [attached file](https://github.com/user-attachments/files/15820398/BUG-.Unpredictable.result.with.overlapping.shapes.zip), I have created two sets of shapes. 
- Both are nearly identical.
- I did not face any warnings/errors during the making (open wire, etc.)
- The only difference is that in one set, the center of the circle is outside the rectangle.
  In the other set, the center of the circle in inside the rectangle.
- I have not trimmed any overlapping edges in the sketch.

But the end-result is totally different:
[image]
Where the center of the circle is inside the rectangle, there are some surface defects (1 and 2).
Also, this solid has a hole in the rectangle.

Note that the circle is not entirely enclosed within the rectangle.

Thus, the user loses control on the shape.

**Desired behavior:**
1. The choice of making a hole (or not) should not be based on the location of the center of the circle (or does this depend on the percentage of circle's area lying outside the rectangle?)
2. If it is breaking Ondsel's rules, then please show some pop up windows that inform the user and let him take an option.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of Ondsel: 64-bit
Version: 2024.2.0dev.37608 (Git)
Build type: Release
Branch: flavor
Hash: cbb758c5dc79fd4c016a9cd45a007d3c7ce1a689
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/India (en_IN)
Installed mods: 
  * Ondsel-Lens 2024.3.26.01
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-06-14:
This is a single body with a single sketch padded and the sketch looks like this : 
[image]

Honestly I'm surprised you get anything when trying to pad this

raindropsfromsky, 2024-06-15:
In fact, I was trying to figure out how the _"Allow multiple solids in Part Design Body by default (experimental)"_ option works.
So I was trying to push the envelope in different ways.

So I tried this, and it worked! :)

closed 2025-02-11 by github-actions[bot]