Ondsel-Development/FreeCAD #91: [1.0 Bughunt] In shaded view mode, sketches disappear

raindropsfromsky, 2024-06-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. In PD WB, press V,6 (or use the _View > Draw style > Shaded_ menu).
2. Create a sketch. Select any plane.
2. Draw any shape (e.g. rectangle)
3. Close the Sketch. The sketch disappears.
4. Press v,1 (or use the _View > Draw style > As is_ menu.) The sketch reappears

### Full version info

```shell
OS: Windows 11 build 22631
Word size of Ondsel: 64-bit
Version: 2024.2.1.37226 (Git)
Build type: Release
Branch: (HEAD detached at 2024.2.1)
Hash: 1a1bd7290494aef925a026c87b808ac58d53c3b0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/India (en_IN)
Installed mods: 
  * 3DfindIT 1.2.0
  * Ondsel-Lens 2024.3.26.01
  * SearchBar 1.0.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

lt72884, 2024-06-15:
I can confirm this as well in sketcher workbench, and part design

raindropsfromsky, 2024-06-16:
> I can confirm this as well in sketcher workbench, part workbench and part design

I can see the disappearance only in the PD WB.
When I am in the Sketcher WB, the sketch does _not_ disappear.

I tried three possible workflows, and found no problem in the Sketcher WB:
1. In a new file, switch to Sketcher WB directly. Draw a sketch. Press v,6.
2. In a new file, start with PD WB, and then create a sketch. (Now we are in the Sketcher WB.) Draw a sketch. Press v,6.
3. I already have a 3D object in the PD WB. double-click on the **Sketch** node in the tree panel, the sketch opens normally, and it is fully visible.

I cannot confirm a problem in the Part WB.
While using the Part WB, I could not add a sketch.
I created a 3D object in the PD WB, which remains fully visible if I press v,6.

sliptonic, 2024-07-20:
The shaded view removes the display of vertices and edges.  Since a sketch is exclusively vertices and edges, there's nothing to show.  This is correct behavior.

closed 2024-07-20 by sliptonic