Ondsel-Development/FreeCAD #101: Malformed unicode characters in Sketcher Expression editor (see screenshot)

ekenberg, 2024-06-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

**Sketcher**: Named constraints with **unicode characters** (in this case Swedish åäö) work well, except for suggested completions in the **Expression editor**. Please see _attached screenshot_. Using one of the suggested paths results in an error. However, typing the correct path manually works as expected.

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of Ondsel: 64-bit
Version: 2024.2.1.37226 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 2024.2.1)
Hash: 1a1bd7290494aef925a026c87b808ac58d53c3b0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.5.24
  * Ondsel-Lens 2024.5.6.01
```

### Subproject(s) affected?

Sketcher

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

prokoudine, 2024-06-30:
This is likely an upstream bug.

@PaddleStroke, @pieterhijma  

#101, #102 amd #105 seem related to each other, at least the first two?

PaddleStroke, 2024-07-01:
I also think it is probably a upstream bug. Unless it's a change by @pieterhijma ?

pieterhijma, 2024-07-01:
In my opinion, (as commented [here](https://github.com/Ondsel-Development/FreeCAD/issues/105#issuecomment-2200533010), this is an upstream bug. Sketcher optional names for constraints should not allow unicode characters, because they are effectively identifiers.

@ekenberg, I recommend creating an issue upstream. I can do that as well if you prefer.

ekenberg, 2024-07-01:
@pieterhijma if you wouldn't mind creating the upstream issue, I think that might be more efficient. If there are any questions or arguments, you will be more suited to answering. Thanks a lot for looking into this.

adrianinsaval, 2024-09-03:
upstream issue: https://github.com/FreeCAD/FreeCAD/issues/5750, it was created for spaces but it's the same for all unsupported characters

closed 2025-02-12 by github-actions[bot]