Ondsel-Development/FreeCAD #103: FCCircularText: No module named 'WebGui'

ekenberg, 2024-06-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Trying to use [the macro FCCircularText](https://wiki.freecad.org/Macro_FCCircularText) results in the error below. Is it necessary to install external packages or should WebGui be included in the AppImage?

`23:15:50  pyException: Traceback (most recent call last):
  File "/home/johan/.local/share/Ondsel/Macro/Macro_FCCircularText.FCMacro", line 83, in <module>
    import WebGui
  File "/tmp/.mount_OndselQdPniW/usr/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ModuleNotFoundError'>: No module named 'WebGui'`

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of Ondsel: 64-bit
Version: 2024.2.1.37226 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 2024.2.1)
Hash: 1a1bd7290494aef925a026c87b808ac58d53c3b0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.5.24
  * Ondsel-Lens 2024.5.6.01
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ekenberg, 2024-06-26:
Perhaps unrelated but: Opening Addon Manager and clicking FCCircularText results in an error:
`Failed to download data from https://wiki.freecadweb.org/Macro_FCCircularText -- received response code 0.`

It is an encryption error, Report View states:
`Error with encrypted connection:[<PySide2.QtNetwork."The certificate has expired" at 0x72b8ec7351c0>]`

However, Chrome shows the certificate as valid:
[image]

and so does [trackssl.com](https://app.trackssl.com/users/sign_up?domain=https%3A%2F%2Fwiki.freecad.org%2FMacro_FCCircularText):
[image]

prokoudine, 2024-06-30:
WebGui is gone from both FreeCAD 0.22dev and Ondsel ES, this macro will have to be updated by the developer. Sorry, but closing

closed 2024-06-30 by prokoudine