Ondsel-Development/FreeCAD #105: VarSet : Unable to create a title in another language

free777cat, 2024-06-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It is not possible to create a name in Ukrainian
[image]

### Full version info

```shell
OS: Linux Mint 22 (MATE/mate)
Word size of Ondsel: 64-bit
Version: 2024.2.0dev.37914 (Git) AppImage
Build type: Release
Branch: flavor
Hash: 88e76e1d516654ee4a020cf4ad676853ca4d2f31
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * Ondsel-Lens 2024.1.22.01
  * sheetmetal 0.4.20
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

prokoudine, 2024-06-30:
@pieterhijma Are properties ASCII-only in FreeCAD by design?

free777cat, 2024-06-30:
It works here
[image]

It doesn't work here
[image]

free777cat, 2024-06-30:
1. Good ----->   Unicode
2. Bad ----->  ASCII

pieterhijma, 2024-07-01:
Thank you for reporting! Properties can only contain ASCII characters (digits, uppercase and lowercase letters, and underscore). I believe the reason is that they should be used as expressions as well. The optional Sketcher name for the constraints allows unicode characters as input and that seems to me a bug. @free777cat, do you want to create an issue in upstream FreeCAD? I can do this as well if you prefer.

free777cat, 2024-07-01:
Do you want to remove UNICODE?

free777cat, 2024-07-01:
To be honest, I don't really like that you can't use your own language :((

pieterhijma, 2024-07-01:
@free777cat, sorry, my explanation may be a bit unclear. Let me try to clarify: Currently in FreeCAD and Ondsel ES, properties can only consist of ASCII characters. The Sketcher allows Unicode characters as input for named constraints, but this is not aligned with the restriction in FreeCAD and Ondsel ES that defines that properties can only consist of ASCII characters. In that sense, Sketcher has a bug and it should be resolved. So, Sketcher should not allow unicode characters as input for named properties.

However, I fully understand that you would like to be able to create properties with unicode characters. This would not be a bug, but a feature request. I don't think it is very difficult to support this in FreeCAD and I would invite you to create an issue in FreeCAD asking for the feature to support unicode as properties.

free777cat, 2024-07-02:
I will be grateful if you create this "issue".

sliptonic, 2024-07-19:
My read of the discussion is that there are two separate issues.  One is a potential bug with sketches.  The other is a feature request.
@pieterhijma let's open the bug report upstream.
@free777cat  If you want to pursue the feature request, that one should be opened in the FreeCAD/FreeCAD tracker.

I don't see anything else to do with this issue here.  Closing.

closed 2024-07-19 by sliptonic