Ondsel-Development/FreeCAD #106: Bughunt-1.0 Incorrect behavior during measurements

free777cat, 2024-06-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[https://youtu.be/kqhQyxn5SJ8]
Incorrect behavior during measurements

### Full version info

```shell
OS: Linux Mint 22 (MATE/mate)
Word size of Ondsel: 64-bit
Version: 2024.2.0dev.37914 (Git) AppImage
Build type: Release
Branch: flavor
Hash: 88e76e1d516654ee4a020cf4ad676853ca4d2f31
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * Ondsel-Lens 2024.1.22.01
  * sheetmetal 0.4.20
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2024-07-19:
I don't understand.  This appears like correct behavior.  You opened the measurement tool and switched documents.  I believe the current design of the measurement tool is to dismiss when the user intentionally closes it.  

I'm closing the issue to keep tracker clean.  If I misunderstood, please comment with clarification and we can reopen.

closed 2024-07-19 by sliptonic

free777cat, 2024-07-20:
Please watch the video again.
"Measurements" => "Open new file" => "Close Measurements"

The actions lead to the deletion of the contents of the file that was opened