Ondsel-Development/FreeCAD #108: Argument not found error when applying external geometry

Footz355, 2024-07-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After adding external geometry, "argument not found" errors occur. Sketch is constructed on top of a face. To reproduce, create a solid, start sketch on solid face, add external geometry, close sketch - error occurs.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of Ondsel: 64-bit
Version: 2024.2.0dev.38021 (Git)
Build type: Release
Branch: flavor
Hash: a34043c4fa4efa3c871e376a4e1536dcdeb86ebc
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Installed mods: 
  * Ondsel-Lens 2024.5.6.01
```

### Subproject(s) affected?

PartDesign

### Anything else?

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

snelweg, 2024-07-03:
same here, happens in the Freecad version too
No external geometry works anymore, no (sub)shapers.

OS: Linux Mint 21.3 (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git) AppImage
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * OpenTheme 2024.6.22
  * FeedsAndSpeeds 0.5.0
  * Freecad-Built-in-themes-beta 1.2.2
  * PieMenu 1.7.3

Footz355, 2024-07-09:
Still occuring in 2024.3.0dev.
[image]

PaddleStroke, 2024-07-22:
I think this is TNP work issues.
Related : https://github.com/FreeCAD/FreeCAD/issues/15549#issuecomment-2241588849
Maybe this issue is going to be fixed by the mentioned PR.

mariohm1311, 2024-08-04:
Any update here? I believe this issue has been fixed (or at least improved on) on the Freecad weekly builds. Is there any timeline for this fix?

closed 2025-02-12 by github-actions[bot]