Ondsel-Development/FreeCAD #109: Bughunt-1.0 "Part Design" You cannot leave only the new part of the mirror "body"

free777cat, 2024-07-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

You cannot leave only the new part of the mirror "body"
[image]
[image]
So that the result would be like this
[image]

### Full version info

```shell
OS: Linux Mint 22 (MATE/mate)
Word size of Ondsel: 64-bit
Version: 2024.2.0dev.38021 (Git) AppImage
Build type: Release
Branch: flavor
Hash: a34043c4fa4efa3c871e376a4e1536dcdeb86ebc
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * Ondsel-Lens 2024.1.22.01 (Disabled)
  * sheetmetal 0.4.21
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

free777cat, 2024-07-02:
Add switch - "leave only mirror", when "transform body"

furgo16, 2024-07-03:
Original issue: https://github.com/FreeCAD/FreeCAD/issues/14117

sliptonic, 2024-07-19:
Feature request.  Removing the 'Bughunt' from the title.
Also, I believe this functionality is possible currently.  I have fully tried it but you should be able to create an App::Link to the original part.  Then set the scale to -1 and adjust the scale list to control which direction to scale along.  The result should be the link being mirrored.

free777cat, 2024-07-20:
This "way" is very roundabout.
As I understand it, it is quite simple to delete the "basic solid" (when the switch is on - "leave only mirror")

sliptonic, 2024-07-20:
The workaround isn't ideal but AFAIK, it does achieve the goal.  
We have to prioritize where we spend development effort.  I've left the issue open and noted that it's a feature request. IMHO, if the goal can be achieved in some other way, then making a new feature to do it easier should be lower priority.  Of course, if someone wants to implement it, a contribution would be gladly accepted  ;-)

free777cat, 2024-07-20:
Thank you. I realized.
src/Mod/PartDesign/App/FeatureMirrored.cpp
[image]
It seems to me that blocking one line from the list (when the switch is on) will solve the problem very easily.
Although maybe I'm wrong.
On a previous project, I took an almost similar "workaround" and got an error embodied in metal.

free777cat, 2024-10-08:
https://github.com/realthunder/FreeCAD/commit/fc7a6be1f287bf1f618b21748c7abe37e09a7460
realthunder solved this problem