Ondsel-Development/FreeCAD #111: Crashes when entering numerical value for open line offset

RayTracedObserver, 2024-07-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In sketcher, draw at least two unconnected lines (not sure if this is the proper term). Select everything, click offset tool, notice cryptic error message, enter numerical value for the offset in the little numerical box, press enter, watch the app crash.

### Full version info

```shell
While this was noticed in the stable build of onsel I also tried it in the weekly build and it was there as well.

Currently I am back on stable : 

OS: macOS 14.4.1
Word size of Ondsel: 64-bit
Version: 2024.2.1.37226 (Git)
Build type: Release
Branch: (HEAD detached at 2024.2.1)
Hash: 1a1bd7290494aef925a026c87b808ac58d53c3b0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * ThreadProfile 1.90.0
  * Ondsel-Lens 2024.7.5.01
  * IconThemes
  * fasteners 0.5.24
```

### Subproject(s) affected?

None

### Anything else?

Here is a .zip file containing a video of the crash and the file used in the video to trigger the crash.
[ondsel_offset_bug.zip](https://github.com/user-attachments/files/16114543/ondsel_offset_bug.zip)

Another issue is open about the error message but it doesn't mention the crashes : https://github.com/Ondsel-Development/FreeCAD/issues/69

Crash report txt file : 
[MacOS crash report.txt](https://github.com/user-attachments/files/16114551/MacOS.crash.report.txt)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-07-08:
https://github.com/Ondsel-Development/FreeCAD/assets/19984177/2bc086f7-ed4e-4ff0-808b-90983d9e595f

So something bad happens if the wires are not coincident with a constraint.

RayTracedObserver, 2024-07-08:
> https://github.com/Ondsel-Development/FreeCAD/assets/19984177/2bc086f7-ed4e-4ff0-808b-90983d9e595f
> 
> 
> 
> So something bad happens if the wires are not coincident with a constraint.

Yes indeed.

And thanks for the little video response :)

sliptonic, 2024-07-19:
@PaddleStroke Is this issue closed by the PR?  Not sure why it didn't automatically close

closed 2024-07-22 by PaddleStroke