Ondsel-Development/FreeCAD #112: Part of the 3d body disappers when creating a sketch on one of its faces

delhub, 2024-07-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While walking through a tutorial from [FreeCAD Beginner Tutorial #1
](https://www.youtube.com/watch?v=VZ16lty5ADg) I ran across an issue where part of the 3D body disappears upon creating a sketch on one face all body parts "above" that face disappear.  

### Full version info

```shell
OS: Windows 11 build 22631
Word size of Ondsel: 64-bit
Version: 2024.2.0dev.37914 (Git)
Build type: Release
Branch: flavor
Hash: 88e76e1d516654ee4a020cf4ad676853ca4d2f31
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Ondsel-Lens 2024.5.6.01
```

### Subproject(s) affected?

None

### Anything else?

Attached is a file of the 3D object with the face selected to be sketched on.
[image]

When creating a sketch on this face, one is presented with the following:
[image]

Note that the presentation of the XY-plane is rotated 180 degrees (this is strange to me, but may not be germaine to this issue).  However, when pressing the '0' key this view is displayed showing the parts that are missing from the object: 
[image]

Note that this is my 4th attempt at trying to use FreeCAD - still frustrating - (but I'm willing to give it one more shot), hence my attempt to use Ondsel (you folks are on the right track).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

delhub, 2024-07-06:
This issue is partially resolved by unchecking all "Visibility Automation" settings under "Sketcher/Display" preferences - all except for "Show objects used for extend geometry" and "Show objects that the sketch is attached to".  However, clicking the "Apply to existing sketches" made no difference if the sketch already existed.  One had to set the setting, then delete the existing sketch and restart the sketch for the settings to take effect.

BlenderManiac40, 2024-07-06:
Try View Section button, shortcut q s on keyboard to change to whole object view. 
[image]

delhub, 2024-07-08:
Thank you.  I should have found that myself.   I truly don't remember, now, if I had tried that.  I would have thought that after clicking the "Apply to existing sketches" in the preferences it would have fixed the proble.  I do know that I used that today on another project to reveal a sketch that was hidden behind other objects.  So, it works for that.  I'm satisfied with the available options at the moment, so we can close this issue.  Thanks, again.

closed 2024-07-08 by delhub