Ondsel-Development/FreeCAD #113: Bughunt-1.0 "Part Design" Linear Array       preview

free777cat, 2024-07-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

lack of preview interferes with work
[image]
[image]
[image]

now it is not visible

[image]
[image]

### Full version info

```shell
OS: Linux Mint 22 (MATE/mate)
Word size of Ondsel: 64-bit
Version: 2024.2.0dev.38021 (Git) AppImage
Build type: Release
Branch: flavor
Hash: a34043c4fa4efa3c871e376a4e1536dcdeb86ebc
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * Ondsel-Lens 2024.1.22.01 (Disabled)
  * sheetmetal 0.4.21
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

free777cat, 2024-07-07:
[image]

prokoudine, 2024-07-07:
I'm sorry, it's hard to say just from the report what the issue is. Sounds like the "Show preview" checkbox ("Оновити вигляд") is available, but... It doesn't work sometimes? Or is not enabled by default? Could you please elaborate?

free777cat, 2024-07-07:
Unfortunately, there is no flag at all. "Show preview" 

The first two photos are how the "preview" works in "realthunder".

"Update view" is another function
[image]

prokoudine, 2024-07-07:
Thank you, so this is more of a request to port from RT's fork?

free777cat, 2024-07-07:
This is a very convenient visualization. The lack of visualization can probably also be considered a "bug". And how to implement it programmatically, you are more experienced.

sliptonic, 2024-07-19:
Looks like a cool feature.  We can add it to the list of RT features to consider porting to FC.  
@prokoudine please add a link to this issue to the document. 

Not a bughunt issue though since behavior is correct as currently designed.  I'm closing the issue otherwise to keep the tracker clean but we will consider porting RTs feature as time permits.

closed 2024-07-19 by sliptonic