Ondsel-Development/FreeCAD #114: CAM - Prefs massive errors output

spanner888, 2024-07-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Attempting to  Edit - Prefs - CAM - Job Prefs - Post Processor and clicking in any edit field produces large error report.\

That's all, no actual changes to prefs, although errors seem to take a few seconds to show.

Drop down does not seem to trigger.
There might be other ways to trigger, not sure, but above seems reliable way to trigger errors.

`08:59:36  Migrating Start Workbench to Start command... done.
09:00:28  Traceback (most recent call last):
09:00:28    File "/tmp/.mount_FreeCAyYeRI5/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 346, in setProcessorListTooltip
09:00:28      self.setPostProcessorTooltip(self.form.postProcessorList, item.text(), "")
09:00:28    File "/tmp/.mount_FreeCAyYeRI5/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 339, in setPostProcessorTooltip
09:00:28      processor = self.getPostProcessor(name)
09:00:28                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^
09:00:28    File "/tmp/.mount_FreeCAyYeRI5/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 333, in getPostProcessor
09:00:28      processor = PostProcessor.load(name)
09:00:28                  ^^^^^^^^^^^^^^^^^^
09:00:28  AttributeError: type object 'PostProcessor' has no attribute 'load'
09:00:28  Traceback (most recent call last):
09:00:28    File "/tmp/.mount_FreeCAyYeRI5/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 346, in setProcessorListTooltip
09:00:28      self.setPostProcessorTooltip(self.form.postProcessorList, item.text(), "")
09:00:28    File "/tmp/.mount_FreeCAyYeRI5/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 339, in setPostProcessorTooltip
09:00:28      processor = self.getPostProcessor(name)
09:00:28                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^
09:00:28    File "/tmp/.mount_FreeCAyYeRI5/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 333, in getPostProcessor
09:00:28      processor = PostProcessor.load(name)
09:00:28                  ^^^^^^^^^^^^^^^^^^
09:00:28  AttributeError: type object 'PostProcessor' has no attribute 'load'
09:00:28  Traceback (most recent call last):
09:00:28    File "/tmp/.mount_FreeCAyYeRI5/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 346, in setProcessorListTooltip
09:00:28      self.setPostProcessorTooltip(self.form.postProcessorList, item.text(), "")
09:00:28    File "/tmp/.mount_FreeCAyYeRI5/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 339, in setPostProcessorTooltip
09:00:28      processor = self.getPostProcessor(name)
09:00:28                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^
09:00:28    File "/tmp/.mount_FreeCAyYeRI5/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 333, in getPostProcessor
09:00:28      processor = PostProcessor.load(name)
09:00:28                  ^^^^^^^^^^^^^^^^^^
09:00:28  AttributeError: type object 'PostProcessor' has no attribute 'load'
09:00:28  Traceback (most recent call last):
09:00:28    File "/tmp/.mount_FreeCAyYeRI5/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 346, in setProcessorListTooltip
09:00:28      self.setPostProcessorTooltip(self.form.postProcessorList, item.text(), "")
09:00:28    File "/tmp/.mount_FreeCAyYeRI5/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 339, in setPostProcessorTooltip
09:00:28      processor = self.getPostProcessor(name)
09:00:28                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^
09:00:28    File "/tmp/.mount_FreeCAyYeRI5/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 333, in getPostProcessor
09:00:28      processor = PostProcessor.load(name)
09:00:28                  ^^^^^^^^^^^^^^^^^^
09:00:28  AttributeError: type object 'PostProcessor' has no attribute 'load'
09:00:28  Traceback (most recent call last):
09:00:28    File "/tmp/.mount_FreeCAyYeRI5/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 346, in setProcessorListTooltip
09:00:28      self.setPostProcessorTooltip(self.form.postProcessorList, item.text(), "")
09:00:28    File "/tmp/.mount_FreeCAyYeRI5/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 339, in setPostProcessorTooltip
09:00:28      processor = self.getPostProcessor(name)
09:00:28                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^
09:00:28    File "/tmp/.mount_FreeCAyYeRI5/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 333, in getPostProcessor
09:00:28      processor = PostProcessor.load(name)
09:00:28                  ^^^^^^^^^^^^^^^^^^
09:00:28  AttributeError: type object 'PostProcessor' has no attribute 'load'
09:00:29  Traceback (most recent call last):
09:00:29    File "/tmp/.mount_FreeCAyYeRI5/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 346, in setProcessorListTooltip
09:00:29      self.setPostProcessorTooltip(self.form.postProcessorList, item.text(), "")
09:00:29    File "/tmp/.mount_FreeCAyYeRI5/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 339, in setPostProcessorTooltip
09:00:29      processor = self.getPostProcessor(name)
09:00:29                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^
09:00:29    File "/tmp/.mount_FreeCAyYeRI5/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 333, in getPostProcessor
09:00:29      processor = PostProcessor.load(name)
09:00:29                  ^^^^^^^^^^^^^^^^^^
09:00:29  AttributeError: type object 'PostProcessor' has no attribute 'load'
09:00:30  Traceback (most recent call last):
09:00:30    File "/tmp/.mount_FreeCAyYeRI5/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 346, in setProcessorListTooltip
09:00:30      self.setPostProcessorTooltip(self.form.postProcessorList, item.text(), "")
09:00:30    File "/tmp/.mount_FreeCAyYeRI5/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 339, in setPostProcessorTooltip
09:00:30      processor = self.getPostProcessor(name)
09:00:30                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^
09:00:30    File "/tmp/.mount_FreeCAyYeRI5/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 333, in getPostProcessor
09:00:30      processor = PostProcessor.load(name)
09:00:30                  ^^^^^^^^^^^^^^^^^^
09:00:30  AttributeError: type object 'PostProcessor' has no attribute 'load'
09:00:30  Traceback (most recent call last):
09:00:30    File "/tmp/.mount_FreeCAyYeRI5/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 346, in setProcessorListTooltip
09:00:30      self.setPostProcessorTooltip(self.form.postProcessorList, item.text(), "")
09:00:30    File "/tmp/.mount_FreeCAyYeRI5/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 339, in setPostProcessorTooltip
09:00:30      processor = self.getPostProcessor(name)
09:00:30                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^
09:00:30    File "/tmp/.mount_FreeCAyYeRI5/usr/Mod/CAM/Path/Main/Gui/PreferencesJob.py", line 333, in getPostProcessor
09:00:30      processor = PostProcessor.load(name)
09:00:30                  ^^^^^^^^^^^^^^^^^^
09:00:30  AttributeError: type object 'PostProcessor' has no attribute 'load'`

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git) AppImage
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
```

### Subproject(s) affected?

CAM/Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2024-07-19:
I'm unable to duplicate the problem.  However, I saw something very similar under specific conditions.  I had my macro path pointing to another directory on the system.  I had some old experimental posts in that directory.

The path preference system looks for valid post processors in several places including the macro directory.  Is it possible you have old stale post processors someplace in the search path?

I'm going to close the issue since I can't duplicate it but will reopen if we can find a reproducible case.

spanner888, 2024-07-20:
Thanks for looking at this.

Your feedback gave me clues and I think the issue is not as I described, but actually triggered by mouse over the post processor list in prefs.

First up apologies as I only tested this in FreeCAD & forgot to test under Ondsel, which as you noted does not show this error.

In FreeCAD, the error appears to be due to missing 
`PostProcessor.load(name)`
 in Mod/CAM/Path/Post/Processor.py, which is STILL present in Ondsel.

Mouse over the post processor list in prefs triggers  setPostProcessorTooltip, which calls getPostProcessor and then attempts: `processor = PostProcessor.load(name)`

But  load(name) no longer exists (did check FC 19 where still was present), guess recent postprocess or Path to CAM refactoring removed this.

```
    def getPostProcessor(self, name):
        if not name in self.processor:
            processor = PostProcessor.load(name)
            self.processor[name] = processor
            return processor
        return self.processor[name]

    def setPostProcessorTooltip(self, widget, name, default):
        processor = self.getPostProcessor(name)
        if processor.tooltip:
            widget.setToolTip(processor.tooltip)
        else:
            widget.setToolTip(default)
```

I am open to your suggestions on next steps - eg raise issue in FreeCAD, but was looking for your feedback and especially to see if there are other differences between Ondsel and FreeCAD (both recent weekly dev versions), that might need followup.

spanner888, 2024-07-20:
Issue also in:
'[code]
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/Australia (en_AU)
Installed mods: 
  * btl 0.9.9 (Disabled)
[/code]
'

sliptonic, 2024-07-20:
confirmed.

sliptonic, 2024-07-20:
https://github.com/FreeCAD/FreeCAD/pull/15537

closed 2024-07-22 by sliptonic