Ondsel-Development/FreeCAD #115: CAM - Tool Bit Library Editor - Save As

spanner888, 2024-07-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Downloaded above FC
Deleted ~/.local/share/FreeCAD/
Deleted ~/.config/FreeCAD/
Ran FC, opened report view
Open CAM WB
Create new document
Open CAM - Tool Bit Library Editor 
In the frist time Tool wizard, create directory to setup Tools & accept all dialogs
In the Tool Bit Library Editor, click on Default library on LHS
Then click "Save As" icon on lower left.
Enter name and click Save, error appears in console:

`09:48:28  Traceback (most recent call last):
09:48:28    File "/tmp/.mount_FreeCA2SSbNe/usr/Mod/CAM/Path/Tool/Gui/BitLibrary.py", line 912, in librarySaveAs
09:48:28      self.updateToolbar()
09:48:28      ^^^^^^^^^^^^^^^^^^
09:48:28  AttributeError: 'ToolBitLibrary' object has no attribute 'updateToolbar'`

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git) AppImage
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
```

### Subproject(s) affected?

CAM/Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

spanner888, 2024-07-08:
Also confirmed in 
`OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of Ondsel: 64-bit
Version: 2024.2.1.37226 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 2024.2.1)
Hash: 1a1bd7290494aef925a026c87b808ac58d53c3b0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Installed mods: 
  * Ondsel-Lens 2024.7.5.02
`

sliptonic, 2024-07-19:
https://github.com/FreeCAD/FreeCAD/pull/15537

sliptonic, 2024-07-22:
merged