Ondsel-Development/FreeCAD #116: Stock Material property is deprecated error after close/reopen simple CAM file.

spanner888, 2024-07-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Create New Doc
Select Part WB
Add Cube
Select CAM WB
Select Cube in tree
Create Job, accept all dialogs
Save file
Close & reopen FC, open file & confusing error occurs.
Error is confusing, because have just created a simple FC solid and CAM-Job that does nothing, but at least NO errors shown.
Now FC gives errors simply after closing & reopening file!

`Stock Material property is deprecated. Removing the Material property. Please use native material system to assign a ShapeMaterial`

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git) AppImage
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
```

### Subproject(s) affected?

CAM/Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

spanner888, 2024-07-08:
Also confirmed in 
`OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of Ondsel: 64-bit
Version: 2024.2.1.37226 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 2024.2.1)
Hash: 1a1bd7290494aef925a026c87b808ac58d53c3b0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Installed mods: 
  * Ondsel-Lens 2024.7.5.02
`

sliptonic, 2024-07-19:
Confirmed.  Converting the stock object to use the new material system, we added logic to remove the old stock material property.  But...we didn't remove the logic that ADDED the property.  DOH!

PR being prepared.  The message will still appear when opening legacy jobs that have the old property.  This is appropriate.

sliptonic, 2024-07-19:
https://github.com/FreeCAD/FreeCAD/pull/15537

sliptonic, 2024-07-22:
merged

closed 2024-07-22 by sliptonic