Ondsel-Development/FreeCAD #118: Crash when trying to create an exploded view

fronald, 2024-07-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When opening the assembly workbench and trying to create an exploded view, the software aborts with a segmentation fault.

### Full version info

```shell
OS: Fedora Linux 40 (KDE Plasma) (KDE/plasma)
Word size of Ondsel: 64-bit
Version: 2024.2.2.37240 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 2024.2.2)
Hash: fbb794cd2e56fba5fd911c8832332b5e5e235319
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Brazil (pt_BR)
Installed mods: 
  * BIM 2021.12.0
  * Ondsel-Lens 2024.7.2.01
```

### Subproject(s) affected?

Assembly

### Anything else?

Video with reproduction steps:

https://github.com/user-attachments/assets/85690215-b5de-4c7d-a1fa-65ae4c96e6cc

Error terminal output:

```
Ondsel 2024.2.2, Libs: 2024.2.2R37240 (Git)
© FreeCAD team and Ondsel Inc., 2001-2024
Ondsel ES is based on the FreeCAD project and released under the terms of the LGPL2.1+ License.
Ondsel ES wouldn't be possible without the FreeCAD community.
      ###          #           #    ##### #####
     #   #         #           #    #     #
     #   # ###   ### #### #### #    #     #
     #   # #  # #  # #    #    #    ####  #####
     #   # #  # #  # #### ###  #    #         #
     #   # #  # #  #    # #    #    #         #
      ###  #  # #### #### #### ##   ##### #####

(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
During initialization the error "'WorkspaceView' object has no attribute 'setFrameUpdate'" occurred in /home/fronald/.local/share/Ondsel/Mod/Ondsel-Lens/./InitGui.py
Please look into the log file for further information
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x40d00) [0x7fe2bae4fd00]
#1  0x7fe2ba132510 in SoDragger::addFinishCallback(void (*)(void*, SoDragger*), void*) from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/./libCoin.so.80+0
#2  0x7fe2bdb2b455 in Gui::View3DInventorPy::addDraggerCallback(Py::Tuple const&) from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libFreeCADGui.so+0x3b5
#3  /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libFreeCADGui.so(+0xa6a213) [0x7fe2bdc6a213]
#4  0x7fe2bdb2361f in Gui::View3DInventorPy::method_varargs_ext_handler(_object*, _object*) from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libFreeCADGui.so+0xf
#5  /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x7fe2bce06068]
#6  /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7fe2bcde5b5c]
#7  /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7fe2bcdf1532]
#8  /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x181) [0x7fe2bce16b71]
#9  /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libpython3.11.so.1.0(+0x21e152) [0x7fe2bce1e152]
#10  /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x24b) [0x7fe2bcde5b3b]
#11  /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7fe2bcdf1532]
#12  /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libpython3.11.so.1.0(+0x235df4) [0x7fe2bce35df4]
#13  /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libpython3.11.so.1.0(+0x2357b0) [0x7fe2bce357b0]
#14  /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libFreeCADBase.so(PP_Run_Method+0xe3) [0x7fe2bc1dd8c3]
#15  0x7fe2bc12920b in Base::InterpreterSingleton::runMethodVoid(_object*, char const*) from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libFreeCADBase.so+0x2b
#16  0x7fe2bd788293 in Gui::PythonCommand::activated(int) from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libFreeCADGui.so+0x53
#17  0x7fe2bd788928 in Gui::Command::_invoke(int, bool) from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libFreeCADGui.so+0x2a8
#18  0x7fe2bd788ce4 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libFreeCADGui.so+0x134
#19  /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7fe2baaeb6f2]
#20  0x7fe2bb96a462 in QAction::triggered(bool) from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libQt5Widgets.so.5+0x32
#21  0x7fe2bb96ce2b in QAction::activate(QAction::ActionEvent) from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libQt5Widgets.so.5+0xab
#22  /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libQt5Widgets.so.5(+0x25d29a) [0x7fe2bba5d29a]
#23  0x7fe2bba5d3d7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libQt5Widgets.so.5+0xe7
#24  0x7fe2bbb4610a in QToolButton::mouseReleaseEvent(QMouseEvent*) from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libQt5Widgets.so.5+0xa
#25  0x7fe2bb9aec08 in QWidget::event(QEvent*) from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libQt5Widgets.so.5+0x178
#26  0x7fe2bb9703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#27  0x7fe2bb978262 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libQt5Widgets.so.5+0x13a2
#28  0x7fe2bd733179 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libFreeCADGui.so+0x99
#29  0x7fe2baab6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libQt5Core.so.5+0x108
#30  0x7fe2bb9763eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#31  /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e6f) [0x7fe2bb9c6e6f]
#32  /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x7fe2bb9ca0c9]
#33  0x7fe2bb9703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#34  0x7fe2bd733179 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libFreeCADGui.so+0x99
#35  0x7fe2baab6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libQt5Core.so.5+0x108
#36  0x7fe2bb1353ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libQt5Gui.so.5+0x84c
#37  0x7fe2bb10b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libQt5Gui.so.5+0xac
#38  /tmp/.mount_OndselkbWM1t/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x7fe2a7d9fbea]
#39  /tmp/.mount_OndselkbWM1t/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7fe2b9310ad7]
#40  /tmp/.mount_OndselkbWM1t/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7fe2b9313ac7]
#41  /tmp/.mount_OndselkbWM1t/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7fe2b93140ce]
#42  0x7fe2bab0aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libQt5Core.so.5+0x66
#43  0x7fe2baab570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libQt5Core.so.5+0x11b
#44  0x7fe2baabd212 in QCoreApplication::exec() from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libQt5Core.so.5+0x82
#45  0x7fe2bd696637 in Gui::Application::runApplication() from /tmp/.mount_OndselkbWM1t/usr/bin/../lib/libFreeCADGui.so+0xa27
#46  /tmp/.mount_OndselkbWM1t/usr/bin/ondsel-es(+0x7688) [0x563894550688]
#47  /lib64/libc.so.6(+0x2a088) [0x7fe2bae39088]
#48  /lib64/libc.so.6(__libc_start_main+0x8b) [0x7fe2bae3914b]
#49  /tmp/.mount_OndselkbWM1t/usr/bin/ondsel-es(+0x79fd) [0x5638945509fd]
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

fronald, 2024-07-22:
Workaround: Passing environment variable `XDG_SESSION_TYPE=X` everything is fine.

PaddleStroke, 2024-07-22:
Hi @fronald 
thanks for the report.
I am not sure what to do about this crash as it seems to be specific to archlinux. Can you tell me a bit more about this XDG_SESSION_TYPE flag and what it does?
The tool internals are being heavily reworked by https://github.com/FreeCAD/FreeCAD/pull/15384 so this may be fixed by this PR.
If after this PR merge you still have this crash, then we'll see.

fronald, 2024-07-22:
I'm really don't know. Without flag same project crashes when creating view exploded on Fedora KDE 6 Wayland with NVidia drivers. Seems like with libcoin?

fronald, 2024-07-22:
Sorry for the confusion. As I need to use this functionality, I did a lot of research and used this flag. When I put it on, everything worked normally. I was happy and commented here. Now I went to record a video with and without this flag and the problem reproduced in both cases. Therefore, false alarm.

Apparently it is intermittent and I had the false feeling that it had been resolved when I inserted the environment variable. But doing more testing, there really is no correlation.

furgo16, 2024-10-02:
Possibly related: https://github.com/FreeCAD/FreeCAD/issues/15706

PaddleStroke, 2024-10-02:
Correct this is the same as https://github.com/FreeCAD/FreeCAD/issues/15706. So I will close this one.

closed 2024-10-02 by PaddleStroke