Ondsel-Development/FreeCAD #121: Assembly Limit Issues

stevenwalton, 2024-07-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The issues are apparent when making the following trivial parts: a box with a hole and a cylinder (idk, place a hex on top to make it look like a real part)

[image]

The problem comes with limits. In our case the box is 30mm as well as the cylinder height. We'll create a cylindrical joint to mate the parts. The question comes into how to limit the movement such that the bolt can go into the part and be limited by the bolt head. It appears that the proper way to do this (with this LCS) is to set a `Length min` to `-30mm` and then `Length max` to some large arbitrary number. 

The issue is that one does not always want to bound in both directions. Current options are `(-inf, inf)` (no limits) or `[a,b]`. It would be nice to include `[a,inf)` and `(-inf,b]`. 

What could make this much more powerful is to add an option for a limiting face. I believe this could be done by using the same functionality of the `Measure` tool since we could measure the distance from the LCS origin to that face and that could define our one-sided limit. Or be able to select 2 faces and have those be the limiting factor. I think this works given that the joint automatically mates so we have a consistent reference coordinate. This functionality could really make the assembly process much more natural (i.e. let humans work with geometries instead of needing to take steps to determine numbers and then calculate)

### Full version info

```shell
OS: macOS 14.5
Word size of Ondsel: 64-bit
Version: 2024.2.2.37240 (Git)
Build type: Release
Branch: (HEAD detached at 2024.2.2)
Hash: fbb794cd2e56fba5fd911c8832332b5e5e235319
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * Ondsel-Lens 2024.7.5.02 (Disabled)
  * freecad.gears 1.2.0
  * fasteners 0.5.24
  * lattice2 1.0.0
  * 3D_Printing_Tools
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rotico, 2024-07-17:
Why don't you just select the bottom circle between screw head and thread. A quick (and dirty) test:
[image]
Seems functional:
[image]
(Win pre-release 2024-07-16)

PaddleStroke, 2024-07-18:
This was already solved by https://github.com/FreeCAD/FreeCAD/pull/14738
I think your version of ES does not have this.

closed 2024-07-18 by PaddleStroke

stevenwalton, 2024-07-18:
Thanks!

As in Ondsel doesn't have it or the Mac version of Ondsel doesn't? I see that it merged with the main branch of Ondsel 3 weeks ago and I am using the `2024.2.2` version, which was released 2 weeks ago. Was just wondering if it wasn't in this release or the Mac version differs.

PaddleStroke, 2024-07-19:
You're welcome.
I'm not sure in which version this ended up though. @prokoudine any idea?