Ondsel-Development/FreeCAD #123: Questionable color choices

wavexx, 2024-07-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I tested both the current stable and weekly build of ondsel, with default settings. I found the default color choices in the "OpenLight" theme questionable and I'm not sure where to report them. Doing this here in good faith to improve usability.

- Preselection is enabled by default, which is good, but both selection and preselection colors are the same making impossible to tell if the one I'm hovering on has already been selected, forcing me to constantly move away from the target just to check. This results in madness when selecting multiple things.

- The color is also identical(?) to the "external geometry" element in the sketcher. Even though the external element is dashed, the problem remains.

### Full version info

```shell
Ondsel_ES_weekly-builds-38021-Linux-x86_64 and Ondsel_ES_2024.2.2.37240-Linux-x86_64
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

prokoudine, 2024-07-18:
The upstream UI theme project is here: https://github.com/obelisk79/OpenTheme/,

@obelisk79, should this be reposted in your tracker?

obelisk79, 2024-07-18:
No, pre-selection and selection colors are the same on purpose. The program, sketcher particularly, uses so many colors needlessly that there are tons of conflict and color management has become highly onerous as a result. While I'm absolutely open to suggestions, I'm not actively seeking to change the current state without a sound proposal. There are many compromises in the themes regarding color I had to make.

wavexx, 2024-07-18:
Color choices are hard, but when you're already fighting with "picking" to work on the right edge, having the same color as selection makes it downright confusing. I cannot tell how many times I wasn't sure if I picked a feature correctly or not with the current behavior before writing this.

Overall I like the color scheme of the sketcher compared to freecad's defaults (which I usually change right away).
I tried changing preselection to a brighter shade of the same purple color at the moment.

closed 2025-02-12 by github-actions[bot]