Ondsel-Development/FreeCAD #125: [1.0 Bughunt] part design,   mirror,     refine

free777cat, 2024-07-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

part design,   mirror
Incorrect operation of "refine"

https://youtu.be/Er2WaB13Ok8

### Full version info

```shell
OS: Linux Mint 22 (MATE/mate)
Word size of Ondsel: 64-bit
Version: 2024.2.0dev.38021 (Git) AppImage
Build type: Release
Branch: flavor
Hash: a34043c4fa4efa3c871e376a4e1536dcdeb86ebc
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * Ondsel-Lens 2024.1.22.01 (Disabled)
  * sheetmetal 0.4.21
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2024-07-29:
I can't duplicate this but it might be something specific to your file.  Please attach a sample file.

free777cat, 2024-07-30:
Very strange. I could not reproduce this bug with the saved file.

closed 2024-07-30 by free777cat