Ondsel-Development/FreeCAD #126: [1.0 Bughunt] When opening the "Assembly" file again (restarting) - the tools are not active

free777cat, 2024-07-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When opening the "Assembly" file again (restarting) - the tools are not active

https://youtu.be/IoIXIeNn1bE

### Full version info

```shell
OS: Linux Mint 22 (MATE/mate)
Word size of Ondsel: 64-bit
Version: 2024.3.0dev.38021 (Git) AppImage
Build type: Release
Branch: flavor
Hash: a34043c4fa4efa3c871e376a4e1536dcdeb86ebc
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * Assembly4 0.50.15
  * Ondsel-Lens 2024.7.5.02
  * sheetmetal 0.4.21
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2024-07-29:
This is correct behavior.  When the document is open, no assembly is active so the tools are disabled.  Activate an assembly and they will enable.

Closing issue as there's nothing to do.

closed 2024-07-29 by sliptonic

free777cat, 2024-07-30:
I consider this a "hidden" behavior of the tools.
It put me in a stupor. There should be a "prompt" or automatic activation of the first "assembly".