Ondsel-Development/FreeCAD #127: [1.0 Bughunt] The "joints" associated with the deleted object are not deleted

free777cat, 2024-07-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

https://youtu.be/IoIXIeNn1bE

The "joints" associated with the deleted object are not deleted

### Full version info

```shell
OS: Linux Mint 22 (MATE/mate)
Word size of Ondsel: 64-bit
Version: 2024.3.0dev.38021 (Git) AppImage
Build type: Release
Branch: flavor
Hash: a34043c4fa4efa3c871e376a4e1536dcdeb86ebc
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * Assembly4 0.50.15
  * Ondsel-Lens 2024.7.5.02
  * sheetmetal 0.4.21
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2024-07-29:
@PaddleStroke  Should they be?  
I can make a case either way.  I think the user will expect joints for deleted objects to also be deleted.

sliptonic, 2024-07-29:
Editing the label.  This is a question of desirable behavior and doesn't result in red ink in the console.

free777cat, 2024-07-30:
Perhaps the last one was written for #124?

PaddleStroke, 2024-07-30:
> to @PaddleStroke <https://github.com/PaddleStroke>.

They should be deleted. Are you using weekly builds?
If so this is a regression

free777cat, 2024-07-30:
v 2024.2.2 = deleted
does not delete in weekly builds

closed 2024-12-06 by PaddleStroke