Ondsel-Development/FreeCAD #129: Movement Gizmo for Imported files is unusable

MrDiver, 2024-08-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
When importing any kind of file the gizmo for the transform is just unusable. You should just add different modes like in Blender that you can toggle if you want to have global, local rotation. But the gizmo should definitely be on the center of the object and it doesn't really make sense to display it elsewhere except for rotation. It's just super annoying to have to deal with this and not even have a reset option.

### Full version info

```shell
.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

prokoudine, 2024-08-17:
This looks a lot like https://github.com/FreeCAD/FreeCAD/issues/12451, what do you think @PaddleStroke ?

PaddleStroke, 2024-08-18:
Yes you are right this is the same issue.
The assembly dragger in the explode view tool has this feature already.

PaddleStroke, 2024-08-18:
I'll close this one as upstream has the issue already

closed 2024-08-18 by PaddleStroke