Ondsel-Development/FreeCAD #137: Sheet metal - workbench lack basic tools

mno13, 2024-09-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Sheet metal workbench lacks basic tools and requires to switch back and forth beetween Sheet Metal and Part Design.

Basic missing tools:
- Create Sketch
- Create Datums
- Create Pad (required for Emboss ("Forming in Wall")
- Create Pockets and holes (thru all by default)
- What about body? Do SM needs body? Or special type of body that allows only SM features? Currently it's chaotic

### Full version info

```shell
OS: Windows 11 build 22631
Word size of Ondsel: 64-bit
Version: 2024.3.0dev.38021 (Git)
Build type: Release
Branch: flavor
Hash: a34043c4fa4efa3c871e376a4e1536dcdeb86ebc
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Installed mods: 
  * Ondsel-Lens 2024.7.5.02
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

prokoudine, 2024-09-06:
What software are you switching from? SW?

mno13, 2024-09-07:
Mostly Solid Edge, but I'm using also CATIA & Inventor for client specific projects. I was also using before some other packages, and AFAIK most of the time their "workbench equivalent" is fully equipped for their role, even if this means "duplicated" tools.

closed 2025-02-12 by github-actions[bot]