Ondsel-Development/FreeCAD #147: 4k display support

InfernalWolf666, 2024-12-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When setting the dimensions of the part, it is impossible to see the numbers

### Full version info

```shell
news
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

InfernalWolf666, 2024-12-01:
When setting the dimensions of the part, it is impossible to see the numbers

prokoudine, 2024-12-04:
Hi, Ondsel is [shutting down](https://ondsel.com/blog/goodbye/), so there will be no further development of Ondsel ES. We recommend using FreeCAD 1.0. It has 99.5% of our patches, and you can easily [make it look like Ondsel ES](https://librearts.org/2024/11/freecad-to-ondsel-es-look/).

closed 2024-12-04 by prokoudine

InfernalWolf666, 2024-12-04:
It's a pity, you had an interesting style.